Klipper3d/klipper #258: Request to add bed skew compensation

leungtech, 2018-03-25:
In marlin, there is a very important function--bed skew compensation, it can adjust skew printer, pls add it, thx!!!

ear0wax, 2018-03-28:
Bed tilt compensation is outlined in klipper/config/example-extras.cfg  

`# Bed tilt compensation. One may define a [bed_tilt] config section to
# enable move transformations that account for a tilted bed.
#[bed_tilt]`

leungtech, 2018-03-28:
This function equals bed skew compensation??? How to use??can you explain a little bit?? Thx!!!

leungtech, 2018-03-28:
in marlin, bed skew compensation not equals auto bed level，what does Bed tilt compensation means in kilpper??thx!!

leungtech, 2018-03-30:
Bed tilt compensation equal auto bed level in marlin, not equal bed skew compensation.
Bed skew compensation equals xy skew correction.
Anyway, the auto bed level in klipper is not really auto, klipper just perform a test and generate a xy result, you need to put these parameter into the config file, not really automatic!!! Just like pid!!

KevinOConnor, 2018-04-27:
I can see where this could be useful.  However, it does not appear anyone is currently working on this.  So, I'm closing this for now.

closed 2018-04-27 by KevinOConnor

KevinOConnor, 2019-07-01:
FYI, bed skew was added in PR #1759 .

-Kevin