Klipper3d/klipper #259: LV8729 doesn't work

luxflow, 2018-03-26:
===== Config file =====
[stepper_a]
step_pin = ar54
dir_pin = !ar55
enable_pin = !ar38
step_distance = .0015625
endstop_pin = ^ar2
homing_speed = 30
position_endstop = 291.775742
angle = 209.984729
arm_length = 269.009

[stepper_b]
step_pin = ar60
dir_pin = !ar61
enable_pin = !ar56
step_distance = .0015625
endstop_pin = ^ar15
position_endstop = 292.597408
angle = 330.205626

[stepper_c]
step_pin = ar46
dir_pin = !ar48
enable_pin = !ar62
step_distance = .0015625
endstop_pin = ^ar19
position_endstop = 290.222678
angle = 90.00000

Read 115004.781415: 'G28\n'
gcode state: absolutecoord=True absoluteextrude=True base_position=[0.0, 0.0, 0.0, 0.0] last_position=[0.0, 0.0, 0.0, 0.0] homing_add=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=1.0 speed=25.0
Internal error on command:"G28"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 153, in process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 66, in <lambda>
    func = lambda params: origfunc(self.get_extended_params(params))
  File "/home/pi/klipper/klippy/gcode.py", line 427, in cmd_G28
    homing_state.home_axes(axes)
  File "/home/pi/klipper/klippy/homing.py", line 109, in home_axes
    self.toolhead.get_kinematics().home(self)
  File "/home/pi/klipper/klippy/delta.py", line 103, in home
    homing_state.home(coord, homepos, endstops, homing_speed)
  File "/home/pi/klipper/klippy/homing.py", line 99, in home
    self.homing_move(movepos, endstops, speed)
  File "/home/pi/klipper/klippy/homing.py", line 66, in homing_move
    mcu_endstop.home_wait(move_end_print_time)
  File "/home/pi/klipper/klippy/mcu.py", line 187, in home_wait
    while self._check_busy(eventtime, home_end_time):
  File "/home/pi/klipper/klippy/mcu.py", line 214, in _check_busy
    raise error("MCU is shutdown")
error: MCU is shutdown
Internal error on command:"G28"
Stats 115006.4: gcodein=87 mcu: mcu_awake=0.006 mcu_task_avg=0.000067 mcu_task_stddev=0.000054 bytes_write=1442 bytes_read=3934 bytes_retransmit=9 bytes_invalid=6 send_seq=127 receive_seq=127 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15993676 print_time=12.431 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.4 pwm=0.000 heater_bed: target=0 temp=27.3 pwm=0.000
Stats 115007.4: gcodein=87 mcu: mcu_awake=0.006 mcu_task_avg=0.000067 mcu_task_stddev=0.000054 bytes_write=1448 bytes_read=4035 bytes_retransmit=9 bytes_invalid=6 send_seq=128 receive_seq=128 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15993673 print_time=12.431 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.4 pwm=0.000 heater_bed: target=0 temp=27.1 pwm=0.000
Stats 115008.4: gcodein=87 mcu: mcu_awake=0.010 mcu_task_avg=0.000100 mcu_task_stddev=0.000136 bytes_write=1454 bytes_read=4165 bytes_retransmit=9 bytes_invalid=6 send_seq=129 receive_seq=129 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15993669 print_time=12.431 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.4 pwm=0.000 heater_bed: target=0 temp=27.2 pwm=0.000
Stats 115009.4: gcodein=87 mcu: mcu_awake=0.010 mcu_task_avg=0.000100 mcu_task_stddev=0.000136 bytes_write=1460 bytes_read=4280 bytes_retransmit=9 bytes_invalid=6 send_seq=130 receive_seq=130 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15993656 print_time=12.431 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.4 pwm=0.000 heater_bed: target=0 temp=26.9 pwm=0.000
Stats 115010.4: gcodein=92 mcu: mcu_awake=0.010 mcu_task_avg=0.000100 mcu_task_stddev=0.000136 bytes_write=1466 bytes_read=4381 bytes_retransmit=9 bytes_invalid=6 send_seq=131 receive_seq=131 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15993659 print_time=12.431 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.4 pwm=0.000 heater_bed: target=0 temp=26.9 pwm=0.000

when I use tmc2100, it works great
I used 80 step per mm when since 1/16 microstep and 1/128 interpolation

when I try to use LV8729, it doesn't work
I used 640 step per mm since 1/128 microstep
so step_distance is 0.0015625

CFG1, CFG2, CFG3 are all high meaning lv8729 1/128 operating mode

tried baud rate 256000 doesn't work

megablue, 2018-03-26:
1/128 microstep is too fine for Atmega 2560 to handle, even for Klipper, more so on a delta.

KevinOConnor, 2018-03-26:
@luxflow - you need to attach the full log (not cut-and-paste portions of it) as described at:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

@megablue - acceptable microstep size is too dependent on other factors to determine if that is the issue - only way to know is with the full log.

3DSTB, 2018-03-27:
I'm not sure if this is the problem: for LV8729 enable is high (not low)

luxflow, 2018-03-27:
```
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.5.0-225-g99f3c99-dirty'
CPU: 4 core ?
Python: '2.7.9 (default, Aug 13 2016, 17:56:53) \n[GCC 4.9.2]'
Start printer at Tue Mar 27 12:16:49 2018 (1522153009.3 264.4)
===== Config file =====
[stepper_a]
step_pin = ar54
dir_pin = ar55
enable_pin = !ar38
step_distance = .0015625
endstop_pin = ^ar2
homing_speed = 30
position_endstop = 291.675742
angle = 209.984729
arm_length = 269.009

[stepper_b]
step_pin = ar60
dir_pin = ar61
enable_pin = !ar56
step_distance = .0015625
endstop_pin = ^ar15
position_endstop = 292.497408
angle = 330.205626

[stepper_c]
step_pin = ar46
dir_pin = ar48
enable_pin = !ar62
step_distance = .0015625
endstop_pin = ^ar19
position_endstop = 290.122678
angle = 90.00000

[extruder]
step_pin = ar26
dir_pin = !ar28
enable_pin = !ar24
step_distance = .01031
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = ar10
sensor_type = ATC Semitec 104GT-2
sensor_pin = analog13
control = pid
pid_kp = 20.952
pid_ki = 0.884
pid_kd = 124.140
min_temp = 0
max_temp = 250
pressure_advance = 0.09
pressure_advance_lookahead_time = 0

[heater_bed]
heater_pin = ar8
sensor_type = ATC Semitec 104GT-2
sensor_pin = analog14
control = pid
pid_kp = 66.886
pid_ki = 1.225
pid_kd = 912.998
min_temp = 0
max_temp = 130

[fan]
pin = ar45

[... 240 lines cut ...]
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Stats 298.0: gcodein=92 mcu: mcu_awake=0.007 mcu_task_avg=0.000074 mcu_task_stddev=0.000063 bytes_write=1571 bytes_read=6189 bytes_retransmit=9 bytes_invalid=6 send_seq=147 receive_seq=147 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15994093 print_time=31.620 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.5 pwm=0.000 heater_bed: target=0 temp=25.0 pwm=0.000
Stats 299.0: gcodein=92 mcu: mcu_awake=0.007 mcu_task_avg=0.000074 mcu_task_stddev=0.000063 bytes_write=1577 bytes_read=6304 bytes_retransmit=9 bytes_invalid=6 send_seq=148 receive_seq=148 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15994091 print_time=31.620 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.5 pwm=0.000 heater_bed: target=0 temp=25.0 pwm=0.000
Stats 300.0: gcodein=92 mcu: mcu_awake=0.010 mcu_task_avg=0.000103 mcu_task_stddev=0.000134 bytes_write=1583 bytes_read=6420 bytes_retransmit=9 bytes_invalid=6 send_seq=149 receive_seq=149 retransmit_seq=2 srtt=0.002 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15994089 print_time=31.620 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.5 pwm=0.000 heater_bed: target=0 temp=25.0 pwm=0.000
Stats 301.0: gcodein=92 mcu: mcu_awake=0.010 mcu_task_avg=0.000103 mcu_task_stddev=0.000134 bytes_write=1589 bytes_read=6535 bytes_retransmit=9 bytes_invalid=6 send_seq=150 receive_seq=150 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15994087 print_time=31.620 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.5 pwm=0.000 heater_bed: target=0 temp=25.0 pwm=0.000
Stats 302.0: gcodein=97 mcu: mcu_awake=0.010 mcu_task_avg=0.000103 mcu_task_stddev=0.000134 bytes_write=1595 bytes_read=6650 bytes_retransmit=9 bytes_invalid=6 send_seq=151 receive_seq=151 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15994085 print_time=31.620 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.5 pwm=0.000 heater_bed: target=0 temp=25.0 pwm=0.000
Stats 303.0: gcodein=97 mcu: mcu_awake=0.010 mcu_task_avg=0.000103 mcu_task_stddev=0.000134 bytes_write=1601 bytes_read=6751 bytes_retransmit=9 bytes_invalid=6 send_seq=152 receive_seq=152 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15994080 print_time=31.620 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.5 pwm=0.000 heater_bed: target=0 temp=25.0 pwm=0.000
Stats 304.0: gcodein=97 mcu: mcu_awake=0.010 mcu_task_avg=0.000103 mcu_task_stddev=0.000134 bytes_write=1607 bytes_read=6880 bytes_retransmit=9 bytes_invalid=6 send_seq=153 receive_seq=153 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15994074 print_time=31.620 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.5 pwm=0.000 heater_bed: target=0 temp=25.0 pwm=0.000
Stats 305.0: gcodein=97 mcu: mcu_awake=0.006 mcu_task_avg=0.000072 mcu_task_stddev=0.000061 bytes_write=1613 bytes_read=6994 bytes_retransmit=9 bytes_invalid=6 send_seq=154 receive_seq=154 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15994078 print_time=31.620 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.5 pwm=0.000 heater_bed: target=0 temp=25.0 pwm=0.000
Stats 306.0: gcodein=97 mcu: mcu_awake=0.006 mcu_task_avg=0.000072 mcu_task_stddev=0.000061 bytes_write=1619 bytes_read=7095 bytes_retransmit=9 bytes_invalid=6 send_seq=155 receive_seq=155 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15994070 print_time=31.620 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.5 pwm=0.000 heater_bed: target=0 temp=25.0 pwm=0.000
Stats 307.0: gcodein=102 mcu: mcu_awake=0.006 mcu_task_avg=0.000072 mcu_task_stddev=0.000061 bytes_write=1625 bytes_read=7224 bytes_retransmit=9 bytes_invalid=6 send_seq=156 receive_seq=156 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready [... cut ...]
Stats 316.0: gcodein=107 mcu: mcu_awake=0.006 mcu_task_avg=0.000073 mcu_task_stddev=0.000061 bytes_write=1679 bytes_read=8243 bytes_retransmit=9 bytes_invalid=6 send_seq=165 receive_seq=165 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15994000 print_time=31.620 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.5 pwm=0.000 heater_bed: target=0 temp=25.0 pwm=0.000
```
weird thing founded by mistake is if I disconnect stepper c endstop, both stepper a,b works well with same setting but when I connect, all stepper doesn't work thanks

KevinOConnor, 2018-03-27:
You are indeed hitting the limits of the atmega2560.  If you want to continue anyway, you can reduce your maximum velocities - try: homing_speed=15, max_z_velocity=30, max_velocity=45

-Kevin

hg42, 2018-03-27:
the power of klipper is this:

you can easily add more MCUs (e.g. an extra RAMPS or one of these pure stepper boards for laser cutters) and give them different tasks.
I once used LV8729 with klipper (either with 128usteps or 64usteps), but I don't remember in which exact configuration.
Either one RAMPS only ran X and Y or even X and Y on different MCUs.

But it wasn't a delta but a corexy. While the MCU doesn't calculate delta kinematics it still has to continuously process all those little segments.
May be this makes a difference here.

Oh, and I forgot the problem with the end stops. I think this prevents distributing the three steppers on separate MCUs.
This is because each stepper must react on all three endstops. And there is no communication between MCUS about end stops.

For corexy I solved this by coupling both end stops electrically (and adding a feature to klipper).

For deltas this probably would be more complicated.
My idea for a solution was to connect each end stop via a diode to it's own input and via another diode to a common input which indicates that any of the end stops is triggered.
So homing would step all steppers until the common input is triggered and then each axis until it's own input is triggered.
Because my corexy replaced the delta I couldn't try the idea.

I am not sure if this could be accomplished by pure software.
The MCU would need a fast channel either to other MCUs or to the host.
Such a common end stop input could also be implemented as an open collector line and pulled down by software, which would make it more flexible, because non moving steppers could be ignored. This would simplify the electrical circuit (no diodes necessary) and the software (the stepper always looks at the same [common] end stop line).

hg42, 2018-03-27:
btw. you could also try 64usteps which should be enough for precision and silence.

KevinOConnor, 2018-03-27:
On Tue, Mar 27, 2018 at 01:51:46PM -0700, Harald wrote:
> Oh, and I forgot the problem with the end stops. I think this prevents distributing the three steppers on separate MCUs.

A delta shouldn't have this problem, as each endstop only stops one
stepper.  Just make sure both the stepper and endstop for a tower is
wired to the same MCU.

It's more complicated for the z-probe though.  In that case, the probe
needs to stop all three towers.

> My idea for a solution was to connect each end stop via a diode to it's own input and via another diode to a common input which indicates that any of the end stops is triggered.
> So homing would step [...]

I think a simpler solution would be to run the endstop wire in
parallel to multiple MCUs.  Specifically, connect one side of the
endstop to a pin on each MCU and connect the other side of the endstop
to ground - all MCUs will see the line pulled up when not triggered
and all will see the pin pulled down when it is triggered.  This has
the advantage of not requiring any special hardware and it doesn't
requiring any special code in the micro-controller.  It would require
a small amount of code in the host software, but I can code that up if
anyone wants to test it.

The above should work for both the probe and for endstops on corexy.

-Kevin

hg42, 2018-03-28:
> A delta shouldn't have this problem, as each endstop only stops one stepper.

oh, nice, I thought it must be more complicated with a delta :-)

Also, nice idea to wire each end stop to each MCU (if it is involved).

I would be happy to test this.
However I am bound to cruwaller's fork (and my own changes), so I would merge this change to my local code. The merge may also conflict with other changes.
So, this might not be sufficient as a test from your point of view.

luxflow, 2018-03-28:
I'm leaving notes for anyone who interesting
3 lv8729 1/64 works well with anycubic linear plus delta, cannot handle 120mm/s but it can handle 80mm/s 
don't know exact limit speed.

In my opinion, lv8729 1/64 less noisy than  tmc2100 1/256

hg42, 2018-03-28:
Really tmc2100 in StealthChop mode? because the other mode is quite noisy (hmm, I have tmc2130 but should be similar for non-SPI modes).
My TMC2130 is nearly inaudible (even with 16usteps interpolated to 256). Though, other sounds from the printer construction are still audible, but not the steps.
But I agree, LV8729 with 64usteps is very silent. And 128 usteps is not worth a limit in speed.

According to speeds of printers, I think this is very much dependent on the parts you print, the materials, the extruder models etc.

I can confirm Kevin's announced speed limit for a single RAMPS board (or similar) with cartesian kinematics.
It can run at nearly 500 mm/s with the usual 80 steps/mm setup.
But no real printer will run that fast, because of other factors.

FYI: my LPC1768 smoothieboard can even run steppers with XY cartesian kinematics smoothly at 2000 mm/s and accelerations up to 20000 mm/s^2 (also with 80 steps/mm).
Note that this was tested with a simple square of 100mm x 100mm and also with a real part (a bar with a lot of holes in all directions) and pure motors (with a small arm to show the movement and position). There were no lost steps, the arm positions were fully reproducible even when running for hours.
But these insane speeds don't matter...no extruder or material or mechnical construction will ever catch up with this.

More important is the exact calculation of acceleration ramps and step times. You hear and see the results.

If you want speed, you should also keep an eye on polygon counts. Once I had pauses of about a second each because the polygon rate was too high (a lot of circles). I assume the serial interface could not catch up with this rate, because reducing serial speed made it worse. The model was sliced with way too high precision.
So, always use a realistic precision. I recommend to compare real print results. You can easily overestimate what can be achieved with usual filaments at higher speeds.

KevinOConnor, 2018-03-28:
On Wed, Mar 28, 2018 at 07:37:45AM -0700, Harald wrote:
> If you want speed, you should also keep an eye on polygon counts. Once I had pauses of about a second each because the polygon rate was too high (a lot of circles). I assume the serial interface [...]

That's almost certainly a limit in OctoPrint (or whatever is sending
the g-code to Klipper).  Use a [virtual_sdcard] definition and print
directly from Klipper if that is a concern.  If the serial line
between host and mcu can not keep up then Klipper will raise an error
to inform you of that.  Also, note that the scripts/graphstats.py tool
(as described in docs/Debugging.md) can be used to graph out the
bandwidth and mcu load during a print.

-Kevin

hg42, 2018-03-28:
Thanks for pointing to graphstats, I never tried it and will keep it in mind.

I know, virtual_sdcard can help (but this didn't exist at that time), but I like to adjust the printer while printing. I think this is not possible with virtual_sdcard?

I don't remember 100%, but I think I started with the usual octopi configuration and then switched to the PC to exclude the bottleneck.
Which means I then tested with Printrun on the PC connected via /tmp/printer to klippy which was also running on the PC and contorlling the MCUs via some usb hub.

The problematic layers of the test object were a 80mm x 10mm with 8 x 4mm holes equally distributed. The pauses always started at the third hole. Then I reduced the slicer precision (the same Cura like in octoprint 15.04?) to 0.05 and the problem was gone. I think the precision had been at least ten times higher (= lower values).

But any recorded data was deleted in the mean time. So let's push this to the world of tales.

luxflow, 2018-03-28:
@hg42  I didn't used spread cycle, and as I heard I should have to use spread cycle since stealth chop cannot hold high torque, and high speed isn't it?

hg42, 2018-03-28:
@luxflow stealth chop is silent and the other makes noises that sound cruel to my ears (high pitched). So I will probably never use this mode.
So you found that LB8729 64ustep is more silent than TMC21000 StealthChop interpolated, which doesn't match my experiences.

For me LV8729 with 64 or 128 usteps is nearly silent but my TMC2130 in interpolated StealthChop mode are still more silent.

May be you have another problem with your TMC2100s, or eventually TMC2130 is better than TMC2100?
Are you sure they are interpolating? I think there are several conditions that can prevent this mode from working in an optimal way.
I usually use slow acceleration (1000) but fairly high speeds (e.g. 100-120 for printing and 200 for traveling).
I also use low current settings which matches well with the slow acceleration. may be they switch modes when becoming hot. I heard, in some situations they may even skip steps and take a large step instead, which results in random noise. I think, TMC drivers have very complex behavior.

But I am still not experienced with these drivers. My time goes elsewhere nowadays.

luxflow, 2018-03-28:
oh sorry I have typo in my previous comment,  I used spreadcycle not stealchop, 64 lv8729 is more silent than 128 spreadcycle
and I used stealchop long long time ago, and as I remember it was extremely silent,  so 128 stealchop is more silient than 64 lv8729 (IMHO)

hg42, 2018-03-28:
I agree :-)
They should have used more different names.

If something would force me to use SpreadCycle, I would certainly choose LV8729 instead. They also have better current ratings (if I remember correctly).

closed 2018-04-11 by KevinOConnor

gdachs, 2018-04-19:
So a STM32F103 just for X and Y would be great  on a CoreXY.