Klipper3d/klipper #260: Bed Z probe support warped bed

kevinrocksman, 2018-03-27:
Will Klipper be able to support a warped bed (with a z probe) where for example the center is bowed resulting in the sides being higher and the middle being lower?

dtandersen, 2018-03-27:
It depends how warped the bed is. My bed has a .127mm bump and it's too much for bed tilt calibration.

Eventually Klipper will have mesh leveling, but it isn't being worked on. In the mean time manually level the bed and use the probe strictly as an endstop. Set stepper_z position_endstop and probe z_offset to the same position.

KevinOConnor, 2018-03-28:
On Tue, Mar 27, 2018 at 05:57:02PM +0000, kevinrocksman wrote:
> Will Klipper be able to support a warped bed (with a z probe) where for example the center is bowed resulting in the sides being higher and the middle being lower?

That's typically handled with "mesh bed leveling" (see issue #96).
It's not currently implemented.

That said, mesh bed leveling is most appropriate for when a printer's
rails are slightly bowed.  If your bed is actually bowed, you should
consider fixing it by printing on glass (or similar).

-Kevin

perkmeister, 2018-03-28:
Thanks for this project, it's very appealing, but I have to say +1 for mesh bed levelling! Warped bed, non-linear sagging of x axis, not completely stiff y carriage and other things also contribute.  Unfortunately cheap Prusa i3 clones can be delinquent in these charateristics, and if people don't have an autolevel probe like a BLTouch they're likely to be doing manual mesh levelling. Cheap thin glass is not necessarily completely flat either.  I think not having this is likely to put off a lot of users.

kevinrocksman, 2018-03-29:
with my bed on solid mounts i still have a uneven bed

https://scontent-dfw5-1.xx.fbcdn.net/v/t1.0-9/29695531_10216111276196948_6760252930880536837_n.jpg?_nc_cat=0&oh=aceae335976e8e275766bd9f131bead4&oe=5B6EABF2

KevinOConnor, 2018-03-31:
Indeed, but it looks like regular bed tilt calibration will help significantly.  In any case, lets move this discussion to the existing issue #96.

closed 2018-03-31 by KevinOConnor