Klipper3d/klipper #261: Instal Klipper

set4up, 2018-03-28:
I can not install the Clipper completely. Because of this, the Clipper service does not start

`octoprint@orangepizero:~$ ./klipper/scripts/install-octopi.sh

###### Running apt-get update...
Hit:1 http://security.debian.org stretch/updates InRelease
Hit:2 http://apt.armbian.com stretch InRelease
Ign:3 http://cdn-fastly.deb.debian.org/debian stretch InRelease
Hit:4 http://cdn-fastly.deb.debian.org/debian stretch-updates InRelease
Hit:5 http://cdn-fastly.deb.debian.org/debian stretch-backports InRelease
Hit:6 http://cdn-fastly.deb.debian.org/debian stretch Release
Reading package lists... Done

###### Installing packages...
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'libncurses5-dev' instead of 'libncurses-dev'
avr-libc is already the newest version (1:1.8.0+Atmel3.5.0-1).
avrdude is already the newest version (6.3-2).
binutils-avr is already the newest version (2.26.20160125+Atmel3.5.3-1).
bossa-cli is already the newest version (1.3~20120408-5+b3).
gcc-avr is already the newest version (1:4.9.2+Atmel3.5.3-1).
libffi-dev is already the newest version (3.2.1-6).
libusb-dev is already the newest version (2:0.1.12-30).
libncurses5-dev is already the newest version (6.0+20161126-1+deb9u2).
libnewlib-arm-none-eabi is already the newest version (2.4.0.20160527-2).
python-virtualenv is already the newest version (15.1.0+ds-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

###### Updating python virtual environment...
./klipper/scripts/install-octopi.sh: line 39: /home/octoprint/klippy-env/bin/pip:  No such file or directory
`
`
octoprint@orangepizero:~$ sudo service klipper start
Failed to start klipper.service: Unit klipper.service not found.
`

luxflow, 2018-03-28:
apt-get update && apt-get install python-pip

set4up, 2018-03-28:
root@orangepizero:~# apt-get update && apt-get install python-pip
Hit:1 http://apt.armbian.com stretch InRelease
Hit:2 http://security.debian.org stretch/updates InRelease
Ign:3 http://cdn-fastly.deb.debian.org/debian stretch InRelease
Hit:4 http://cdn-fastly.deb.debian.org/debian stretch-updates InRelease
Hit:5 http://cdn-fastly.deb.debian.org/debian stretch-backports InRelease
Hit:6 http://cdn-fastly.deb.debian.org/debian stretch Release
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-pip is already the newest version (9.0.1-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

set4up, 2018-03-28:
I deleted the Klipper folder and put it on again, but the last error still remains

###### Updating python virtual environment...
./klipper/scripts/install-octopi.sh: line 39: /home/octoprint/klippy-env/bin/pip: No such file or directory

KevinOConnor, 2018-03-28:
On Wed, Mar 28, 2018 at 10:01:31AM -0700, set4up wrote:
> I deleted the Klipper folder and put it on again, but the last error still remains
> 
> ###### Updating python virtual environment...
> ./klipper/scripts/install-octopi.sh: line 39: /home/octoprint/klippy-env/bin/pip: No such file or directory

Try removing the klippy-env directory:

  rm -r /home/octoprint/klippy-env/

And then retry.

-Kevin

set4up, 2018-03-28:
A little better, but other errors in the end. I apply completely

octoprint@orangepizero:~$ rm -rf /home/octoprint/klipper/
octoprint@orangepizero:~$ rm -rf /home/octoprint/klippy-env/
octoprint@orangepizero:~$ git clone https://github.com/KevinOConnor/klipper
Cloning into 'klipper'...
remote: Counting objects: 5709, done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 5709 (delta 5), reused 11 (delta 4), pack-reused 5692
Receiving objects: 100% (5709/5709), 2.07 MiB | 1.91 MiB/s, done.
Resolving deltas: 100% (4234/4234), done.
octoprint@orangepizero:~$ ./klipper/scripts/install-octopi.sh

###### Running apt-get update...
Hit:1 http://apt.armbian.com stretch InRelease
Hit:2 http://security.debian.org stretch/updates InRelease
Ign:3 http://cdn-fastly.deb.debian.org/debian stretch InRelease
Hit:4 http://cdn-fastly.deb.debian.org/debian stretch-updates InRelease
Hit:5 http://cdn-fastly.deb.debian.org/debian stretch-backports InRelease
Hit:6 http://cdn-fastly.deb.debian.org/debian stretch Release
Reading package lists... Done

###### Installing packages...
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'libncurses5-dev' instead of 'libncurses-dev'
avr-libc is already the newest version (1:1.8.0+Atmel3.5.0-1).
avrdude is already the newest version (6.3-2).
binutils-avr is already the newest version (2.26.20160125+Atmel3.5.3-1).
bossa-cli is already the newest version (1.3~20120408-5+b3).
gcc-avr is already the newest version (1:4.9.2+Atmel3.5.3-1).
libffi-dev is already the newest version (3.2.1-6).
libusb-dev is already the newest version (2:0.1.12-30).
libncurses5-dev is already the newest version (6.0+20161126-1+deb9u2).
libnewlib-arm-none-eabi is already the newest version (2.4.0.20160527-2).
python-virtualenv is already the newest version (15.1.0+ds-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

###### Updating python virtual environment...
Running virtualenv with interpreter /usr/bin/python2
New python executable in /home/octoprint/klippy-env/bin/python2
Also creating executable in /home/octoprint/klippy-env/bin/python
Installing setuptools, pkg_resources, pip, wheel...
  Complete output from command /home/octoprint/klippy-env/bin/python2 - setuptools pkg_resources pip wheel:
  Collecting setuptools
Exception:
Traceback (most recent call last):
  File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/index.py", line 792, in get_page
    "Cache-Control": "max-age=600",
[... 6 lines cut ...]
  File "/home/octoprint/klippy-env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/home/octoprint/klippy-env/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py", line 37, in send
    cached_response = self.controller.cached_request(request)
  File "/home/octoprint/klippy-env/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/controller.py", line 111, in cached_request
    resp = self.serializer.loads(request, cache_data)
  File "/home/octoprint/klippy-env/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/serialize.py", line 114, in loads
    return getattr(self, "_loads_v{0}".format(ver))(request, data)
  File "/home/octoprint/klippy-env/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/serialize.py", line 176, in _loads_v2
    cached = json.loads(zlib.decompress(data).decode("utf8"))
error: Error -5 while decompressing data: incomplete or truncated stream
----------------------------------------
...Installing setuptools, pkg_resources, pip, wheel...done.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 2375, in <module>
    main()
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 724, in main
    symlink=options.symlink)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 992, in create_environment
    download=download,
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 922, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 817, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /home/octoprint/klippy-env/bin/python2 - setuptools pkg_resources pip wheel failed with error code 2

set4up, 2018-03-28:
I today already reinstalled the OS

>
>> ###### Updating python virtual environment...
>> Running virtualenv with interpreter /usr/bin/python2
>> New python executable in /home/octoprint/klippy-env/bin/python2
>> Also creating executable in /home/octoprint/klippy-env/bin/python
>> Installing setuptools, pkg_resources, pip, wheel...
>>   Complete output from command /home/octoprint/klippy-env/bin/python2 - setuptools pkg_resources pip wheel:
>>   Collecting setuptools
>> Exception:
>
>Not sure what's going on, but it's definitely not a klipper issue.
>The command being run is:
>
>virtualenv /home/octoprint/klippy-env
>
>And that's a standard python setup command.  You might want to try a
>complete OS reinstall.
>
>-Kevin

С уважением, Евгений

set4up, 2018-03-28:
I already reinstalled the OS today
Thank you for your patience

luxflow, 2018-03-28:
https://github.com/interbrite/letsencrypt-vesta/issues/46#issuecomment-273014451
after I googled your error msg, I got this one,
so try `pip install --upgrade pip`

set4up, 2018-03-29:
Completely reinstalled the wasps and the rest. It seems it helped. service Clipper now starts

set4up, 2018-03-29:
Instructions from here, I changed them a little
http://deloarts.com/en/3d-printing/octoprint-on-orange-pi-zero/
http://3dtoday.ru/blogs/eone/klipper-or-as-anet-a8-gained-a-new-body-and-brain-/

KevinOConnor, 2018-04-03:
Was this resolved?  The issue was with a "pip" parameter outside of the Klipper installation?

closed 2018-04-18 by KevinOConnor

cahayez, 2018-06-30:
I ran into this today. Solved it by running 

`virtualenv /home/octoprint/klippy-env`

then
` ./klipper/scripts/install-octopi.sh`

keepfishes, 2018-09-06:
> I ran into this today. Solved it by running virtualenv /home/octoprint/klippy-env then ./klipper/scripts/install-octopi.sh

Totally worked for me after...