Klipper3d/klipper #262: Request: I'd like to be able to define unique UUIDs to each klipper firmware board

jwcrawley, 2018-03-29:
I plan on running multi-RAMPS configuration on my printer. My configuration is the following;

First board: X, Y, Z motors, thermistors, heated bed, endstops
Second Board: extruders 1-5

The problem currently, is that if I'm using the same electronics for each, the assignment from /dev/ttyUSB[0-9] or /dev/ttyACM[0-9] is purely done on which one was first.

I would like the firmware to emit a UUID in the same way Marlin has a UUID so I can assign on my printer.cfg file these IDs instead of using ephemeral /dev/tty[USB|ACM][0-9]

Is this doable? What would need done to accomplish this?

Sincerely,
CrankyLinuxUser

theopensourcerer, 2018-03-29:
See the FAQ:

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

Hint: use /dev/serial/by-id/

dragonnn, 2018-03-29:
Some linux distribution doesn't have this. But as I find you just need add it to rules.d in udev. More info here https://superuser.com/questions/839667/how-is-by-id-created-on-linux

jwcrawley, 2018-03-29:
Thank you both @theopensourcerer @dragonnn . I wish using /dev/serial/by-id would work, but it doesn't.

I'm using cheap clone-duino megas from China. When I go do a "lsusb -v" and grep for iSerial, its "0" for each board. Legit Arduino Megas do have a iSerial and those aren't an issue, except for the fact that they cost $50 each.

I also looked at using Vid:Pid, but again, that's non-unique between the same hardware.

theopensourcerer, 2018-03-29:
Odd - I'm using a MKS Gen v1.4 which is a very cheap single board RAMPS from China and it works OK with /dev/serial/by-id . 

I'm not at home right now but it sounds more like the Linux end which might the issue to me. Is it a genuine RPi and are you using Raspbian/Octopi?

jwcrawley, 2018-03-29:
Nope. I'm using the OrangePi Nano with Sunxi kernel. 

I'll try one more time when I get home. I'll see if rebooting helps. It'll be in 8 hours before I can test once more.

Thank you

KevinOConnor, 2018-03-29:
On Thu, Mar 29, 2018 at 01:01:59PM +0000, Josh Conway wrote:
> Thank you both @theopensourcerer @dragonnn . I wish using /dev/serial/by-id would work, but it doesn't.
>
> I'm using cheap clone-duino megas from China. When I go do a "lsusb -v" and grep for iSerial, [...]

If /dev/serial/by-id doesn't work, then use /dev/serial/by-path
instead.

-Kevin

jwcrawley, 2018-03-29:
Ok, I just got home and checked this solution. As I suspected, it reports a generic name and port position when more than 1 are plugged in. 

After further digging, it appears the problem is also noted [here](https://arduino.stackexchange.com/questions/6617/setting-serial-number-on-ch340-usb-serial-device) regarding programming an iSerial on a ch340 chip.

And it looks like there's no way to inject a USB iSerial tag. The resulting solution is what you indicated @KevinOConnor , and that's to use the /dev/serial/by-path , and map the devices that way and never move them. 

Thank you, but this seems to be a fool's errand due to me cheaping out with ch340 usb-serial chips.

closed 2018-03-29 by jwcrawley