Klipper3d/klipper #263: What is the step distance of mk8 direct drive?

leungtech, 2018-03-30:
In marlin, the step distance of mk8 direct drive is 93, so by calculation it should be 1/93= 0.0107526882 in klipper. But it doesn't work, the plastic being extruded is too much, much much more than calculation, how to calculate in klipper?? I use a4988 16 microstep, thx!

cmock, 2018-03-30:
Your calculation is correct, and your numbers make sense (my MK8 extruder was at 95 steps/mm). 

I converted two printers over from Marlin with thet type of calculation, and it works fine, so there must be some other reason.

Can you try an extruder calibration (command to extrude 100mm and measure how much was really extruded)?

megablue, 2018-03-30:
it varies from supplier to supplier, some have deeper grove and some don't. but 80~100 steps per mm is the ballpark.

leungtech, 2018-03-30:
I will try it later , I use 64 microstep for x and y, and 16 microstep for z and extruder, maybe that is a problem, does klipper support different microstep for x y z extruder??

hg42, 2018-03-30:
> does klipper support different microstep for x y z extruder??

yes, I have that myself (should be usual)

do you take these Marlin values from the configuration header file or from eeprom settings?

hg42, 2018-03-30:
also, the slicer may have been configured to set some values in the header. These may fail for klipper, because Kevin does not support several gcodes.

KevinOConnor, 2018-04-08:
The initial question looks like it was answered.

closed 2018-04-08 by KevinOConnor