Klipper3d/klipper #265: How to set Auto bed level

leungtech, 2018-03-30:
First of all, is there any auto bed leveling like marlin?

I have tried z probe and bed tilt calibrate, I have added sereval points which is copied from bltoch. Now when I type bed_tilt_calibrate , the print will first perform g28 home all axis. I also write home override so that the printer will home x and y axis first and then go to x100 y100 to perform z homeing by z probe. Then it will go to the 10 points but z will not lower to do auto bed level and go to next point without lower z, what should I do. Any setting wrong? What is the procedure to set auto bed leveling? Thx!

leungtech, 2018-03-30:
ok, done!! thx!!

closed 2018-03-30 by leungtech

hg42, 2018-03-30:
perhaps you should try harder before you open an issue?
We all had to adapt to the possibilities of klipper.

Also, issues are not the right place for questions.
I think the mailing list would be better suited for that.

leungtech, 2018-03-31:
OK sorry to disturb you , as I am not studying software or hardware 
or computer, and there ia too little instructions to setup klipper. I am in a hurry to change all my five 3d printer to klipper. The idea of klipper is so great, but there is too little instructions to follow setup, community support is also not enough at this moment. I hope klipper will have more functions like marlin and more instructions for beginner and eventually more people can use this great firmware. Anyway, sorry to disturb you.

tvercruysse, 2018-12-03:
> perhaps you should try harder before you open an issue?
> We all had to adapt to the possibilities of klipper.
> 
> Also, issues are not the right place for questions.
> I think the mailing list would be better suited for that.

@hg42 
Issues are a **great** place to discuss problems you are having with software (hence the name). I find it quite egocentric of you to just tell someone looking for help to "try harder". He clearly tried, without success. so isn't it to the community to share the knowledge accumulated over the years with others instead of telling them to figure it out by themselves ? 

Klipper is great and the documentation is phenomenally informative in regard of techniques used, but they do indeed lack configuration instructions (that's something an issue like this addresses). And that's okay with me, but please understand that not everyone has the same level of experience.