Klipper3d/klipper #268: Cuurent extruder motion settings and filament types question

w1ebr, 2018-03-31:
Hi Kevin,
The current settings worked amazing well for TPU but with very stiff filament I am seeing a lot more blobs from retraction and repriming. Given what you know about the motion parameters you have set up for the extruder, does that make sense to you )or are they unrelated)?
Thanks!
Gene

KevinOConnor, 2018-03-31:
On Fri, Mar 30, 2018 at 05:15:57PM -0700, w1ebr wrote:
> The current settings worked amazing well for TPU but with very stiff filament I am seeing a lot more blobs from retraction and repriming. Given what you know about the motion parameters you have set up for the extruder, does that make sense to you )or are they unrelated)?

It's not clear if you're using pressure advance or not.  If you are,
you need to tune pressure advance for each new filament.  (For
example, I have two different spools of PLA where one requires a pa
value of 0.07 and the other requires 0.03 on the same printer.)

If you're not using pressure advance, you might want to try it out.

It's not clear if you're using the same retraction settings with both
filaments.  If you're using a large retraction setting, then ooze
during the retraction itself could be an issue.  Pressure advance
helps with this (and also allows one to use a smaller retraction
value).  Otherwise, some people will customize the
max_extrude_only_velocity/accel so as to retract faster.

-Kevin

w1ebr, 2018-03-31:
I didn't even see this! Duh and thanks!

closed 2018-03-31 by w1ebr