Klipper3d/klipper #269: MKS 1.4 and dual Z in Creality 10S

ChrisMechanic, 2018-03-31:
Dear All!
I have Creality 10S and decided to flash klipper not into original motherboard, but to do it on MKS 1.4. However, I didn't check previously, MKS 1.4 does not offer dual Z motor. Or does it?

Here: http://osoyoo.com/2016/06/30/mks-1-4-3d-printer-board-marlin-firmware-installation-guide/
is written, in case of dual Z, I should connect 2nd Z to E1 in MKS 1.4.
Is that correct?
Will klipper work like that?

Thank you in advance for your responses!
[image]

bmc0, 2018-03-31:
I've not actually tested it, but it looks like you can just define a `[stepper_z1]` section and set the pins correctly. Klipper doesn't care what the axis is "supposed" to be. See [example-extras.cfg at line 147](https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L147) and [printer-anycubic-i3-mega-2017.cfg at line 40](https://github.com/KevinOConnor/klipper/blob/master/config/printer-anycubic-i3-mega-2017.cfg#L40).

KenshiHH, 2018-03-31:
`[stepper_z1]
step_pin: ar36
dir_pin: ar34
enable_pin: !ar30
step_distance: 0.00625
endstop_pin: ^ar19`

my config for the mks gen 1.4 with dual z and dual z endstop.
working fine

w1ebr, 2018-03-31:
Hello! Do the two z steppers run independently? If yes, how are they handled in Klipper?
Thanks!

akpch, 2018-04-04:
I just plugged second z into E1 and stop into Z max.

closed 2018-04-11 by KevinOConnor