Klipper3d/klipper #270: Connection problem after printer reboot

niko9797, 2018-03-31:
Hi, first of all thanks for the great job, i discovered klipper some weeks ago and i already love it.

I just have a little issue with connection between the printer board (MKS GEN L v1.0) and the rpi.
every time i turn off the printer, it doesnt matter what i do it will just not reconnect to the rpi untill i restart the octopi server. The MKS board doesnt take power from usb port just from wall. to turn the printer on/off i just plug/unplugg it.

I checked the log and its says doesnt find ttyUSB0, i checked my system and after turning off and on the printer appears as ttyUSB1, not as configured in printer.cfg

Is this just my config problem, are i´m turning it on/off incorrectly, or there is no way to avoid this? Thanks, Niko

[klipperissue.txt](https://github.com/KevinOConnor/klipper/files/1865707/klipperissue.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/1865724/klippy.log)
[printercfg.txt](https://github.com/KevinOConnor/klipper/files/1865725/printercfg.txt)

grigorig, 2018-03-31:
The device name changes because ttyUSB0 was in use when it was disconnected. You simply cannot rely on these automatic device names. There is a good workaround, however. Identify the port by ID:
```
[mcu]
#serial: /dev/ttyUSB0
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
```
Klipper still doesn't automatically recognize the printer (at least for me), but entering a RESTART or RESTART_FIRMWARE command in the Octoprint terminal does the job. There's an addon that allows you to map buttons to GCODE commands which I use to make that more user-friendly.

niko9797, 2018-03-31:
Thank you very much for your fast help,
right now printer is working, but the workaround seems good to me and will test it tomorrow. Will also check the addon as looks very interesting.
Should instructions of the way of setting up the serial port be added somewhere in default setup instructions files, or in printer.cfg files examples maybe as an appointment?

KevinOConnor, 2018-03-31:
FYI, it's documented in the FAQ: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

Alas, the value returned is different on every board, so we can't use it directly in the config examples.

niko9797, 2018-03-31:
Thanks again, i should had check before the FAQ, my fault, respecting the modif of the config files, i think i expressed myself badly, what i was reffering to was to add more info, not to change the value, i attach an example:

Actual example.cfg:
```
# 
...
Micro-controller information.
[mcu]
serial: /dev/ttyACM0
#   The serial port to connect to the MCU. The default is /dev/ttyS0
#baud: 250000
#   The baud rate to use. The default is 250000.
pin_map: arduino
#   This option
....
```
Proposed:
```
...
# Micro-controller information.
[mcu]
serial: /dev/ttyACM0
#   The serial port to connect to the MCU. The default is /dev/ttyS0
#   To find your port, check it by id using ls /dev/serial/by-id
#   For more info look out klipper FAQ
#baud: 250000
#   The baud rate to use. The default is 250000.
pin_map: arduino
#   This option
....
```

If you think its worthy i could try to open a pull request for it tomorrow, if no, my issue is resolved and post can be deleted/closed,
Greetings, Niko

KevinOConnor, 2018-04-20:
That makes sense.  I updated the config comments (commit 832c0bc0).

closed 2018-04-29 by KevinOConnor