Klipper3d/klipper #271: How to interpret step rate benchmarks?

grigorig, 2018-03-31:
I'm working on a port of Klipper to STM32F103 at the moment. Everything actually seems to work better than expected, which is great. Right now I want to benchmark step rates, and I'm a bit at a loss how the quoted step benchmarks in the documentation are achieved.

The step rate on fast MCUs is limited by the UNSTEP_TIME; even if I set the interval to 1 cycle, I don't see any timing errors - and the logical analyzer shows a step rate which indicates the UNSTEP_TIME limitation.

So I assume the numbers are the *sum* of all step rates combined when simultaneously stepping with multiple steppers. Is that correct?

I've been using some crude commands to generate step signals with console.py, e.g.

```
allocate_oids count=1
config_stepper oid=0 step_pin=PC13 dir_pin=PB4 min_stop_interval=0 invert_step=1
finalize_config crc=0
set_next_step_dir oid=0 dir=1
reset_step_clock oid=0 clock={clock + 72000000}
queue_step oid=0 interval=720 count=65000 add=0
```

Is this the way to go?

KevinOConnor, 2018-03-31:
Yes - the benchmarks are the sum of all active steppers.  So, if three steppers are running simultaneously, then I multiply the rate by three.

I calculate the benchmark rates using the console.py tool.  The following are my notes from the Arduino Due test (the tests on the beaglebone and AVR are similar) - I cut and paste the below into the console.py session:

```
# Commands for configuring Arduino Due board
#
allocate_oids count=3
config_stepper oid=0 step_pin=PB27 dir_pin=PA21 min_stop_interval=0 invert_step=0
config_stepper oid=1 step_pin=PB26 dir_pin=PC30 min_stop_interval=0 invert_step=0
config_stepper oid=2 step_pin=PA21 dir_pin=PC30 min_stop_interval=0 invert_step=0
finalize_config crc=0

#### Stepper tests
SET start_clock {clock+freq}
SET ticks 374

reset_step_clock oid=0 clock={start_clock}
set_next_step_dir oid=0 dir=0
queue_step oid=0 interval={ticks} count=60000 add=0
set_next_step_dir oid=0 dir=1
queue_step oid=0 interval=3000 count=1 add=0

reset_step_clock oid=1 clock={start_clock}
set_next_step_dir oid=1 dir=0
queue_step oid=1 interval={ticks} count=60000 add=0
set_next_step_dir oid=1 dir=1
queue_step oid=1 interval=3000 count=1 add=0

reset_step_clock oid=2 clock={start_clock}
set_next_step_dir oid=2 dir=0
queue_step oid=2 interval={ticks} count=60000 add=0
set_next_step_dir oid=2 dir=1
queue_step oid=2 interval=3000 count=1 add=0
```

I bisect the `ticks` parameter until I get to the lowest tick rate that Klipper will regularly run the commands at without returning an error.  (At a rate too fast, Klipper will report a "Rescheduled timer in the past" or "Stepper too far in past" error.)

In the event of an error, I cut-and-paste this into console.py to clear the error:
```
#### Shutdown clearing
clear_shutdown
```

After clearing the error, one can cut-and-paste the first set of commands (starting from the "Stepper tests" section).  For the Due, a tick rate of 374 for three simultaneous steppers would calculate out to 4200000. / 374. * 3. == 336898, which I then round to 337K steps per second.  (The Due timer rate runs at 42Mhz.)

grigorig, 2018-04-06:
The console.py script worked quite good for me. I think this can be closed now.

closed 2018-04-06 by grigorig