Klipper3d/klipper #273: Unknown option "pressure_advance_lookahead_time" in 0.6.0

w1ebr, 2018-04-01:
Recv: // Unknown option 'pressure_advance_lookahead_time' in section 'extruder'
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted
Recv: !! Unknown option 'pressure_advance_lookahead_time' in section 'extruder'
Recv: ok

[klippy.log](https://github.com/KevinOConnor/klipper/files/1866025/klippy.log)

cmock, 2018-04-01:
That is something I also stumbled across: if ```pressure_advance``` is unset or zero, ```pressure_advance_lookahead_time``` must not be present. I think it should be documented more clearly.

w1ebr, 2018-04-01:
I made the edit to extruder.py that I show above - I am waiting to see if Kevin considers what he implemented a bug or not

grotesk86, 2018-04-02:
I think @cmock is right, it just checks if `pressure_advance `is set and if not it's not necessary to have `pressure_advance_lookahead_time ` set.

w1ebr, 2018-04-02:
It also doesn't work if it is set to 0, which is what the pressure advance documentation specifies ( i think) when beginning to determine what the right value should be.

KevinOConnor, 2018-04-03:
On Sun, Apr 01, 2018 at 09:20:31PM +0000, w1ebr wrote:
> I made the edit to extruder.py that I show above - I am waiting to see if Kevin considers what he implemented a bug or not

It doesn't make sense to have a non-zero
pressure_advance_lookahead_time if there is no pressure_advance being
done.  And, the code will not react well (see the lookahead() method)
if that is done.

So, we could read the parameter and require it to be zero, but that
seems counter productive.

-Kevin

w1ebr, 2018-04-03:
Kevin, 
The problem I had was that I used 0.0 for the initial pressure advance value and the code complained that 
pressure_advance_lookahead_time was an unknown option.

KevinOConnor, 2018-04-03:
FYI, Michael addressed this with his SET_PRESSURE_ADVANCE code in commit a3017133.

w1ebr, 2018-04-03:
I made the update - thanks!

closed 2018-04-03 by w1ebr