Klipper3d/klipper #275: Request for feature: Master peed scalar for fan

afoxinsocks, 2018-04-01:
So, I have a 24v PSU and a 12v FAN... temporarily, I've set fan max_power to 0.5, which appears to output the 12v that I need, but I'd like to have a master scalar the fan instead of clipping the maximum to 0.5 and losing the functionality of that setting.

I'll probably implement this myself in the next couple of days, but if you get to it first or have some naming pointers for the configuration value, thanks in advance!

dragonnn, 2018-04-02:
This is not how PWM works! You fan still getting 24V only 50% the time, only the average voltage is 12V. Running it like that will still shorten it life time. I suggest using a step-down converter:
1. Connect the step-down converter directly to 24V
2. Fan positive goes to the step-down output
3. Fan negative goes to the negative PWM output on you board
This way you can still use PWM control on you fan and it is getting true 12V.

KevinOConnor, 2018-04-20:
I agree that it makes more sense to scale the requested fan speed.  I updated the code to do that (commit 85e6cd86).

closed 2018-04-27 by KevinOConnor