Klipper3d/klipper #276: PID_CALIBRATE internal error

afoxinsocks, 2018-04-01:
[pid_calibrate_failure.log](https://github.com/KevinOConnor/klipper/files/1866780/pid_calibrate_failure.log)

Suspect something is slightly bad in the calibration function. Command was the example straight from the config_checks README -- PID_CALIBRATE HEATER=extruder TARGET=170
FWIW, the extruder temp did go up to 170 and then error'd out.

afoxinsocks, 2018-04-02:
This appears to have been caused by the heater verification failing during the PID calibration. After trying a lower temperature, (120 instead of 170), it was successful... don't know if it's my thermistor or heater cartridge at fault, failing to get to 170+ easily.

closed 2018-04-02 by afoxinsocks

schwab, 2018-07-02:
Try thermally isolating your print head from the the mounting hardware if possible. I've found that doing this helps the problem immensely.  However, I will say it can be difficult to accomplish in practice.  In my case metal washers between the thermal head and the metal mounting hardware act as a conduit and transfer a lot of heat into the mounting structure. I've been able to isolate them with a thin layer of teflon or thermal tape, but it doesn't always last long and problems are reoccurring.