Klipper3d/klipper #277: Always "move out of range" problem

leungtech, 2018-04-02:
i use bed tile compensation, all settings are perfect but when it start to print, it print something and then the printer hangs. The terminal said "move out of range ". What is the range stands for?? The z-offset and stepper z endstop is same number.

Send: N74 G1 X143.333 Y128.999 E0.00859*82
Recv: ok
Send: N75 G1 X55.635 Y128.959 E2.91685*111
Recv: !! Move out of range: 55.635 128.959 -0.136 [21.074]
Changing monitoring state from 'Printing' to 'Operational'
Send: M400
Send: M114
Recv: ok
Recv: ok
Recv: X:143.333 Y:128.999 Z:0.200 E:5.657
Send: M84
Send: M104 T0 S0
Send: M140 S0
Send: M106 S0

leungtech, 2018-04-02:
when i restart the printer, and perform g28, the printer is normal, but after g28, i type g1 x20, it shows move out of range again, but g1 x30 is ok, what is the range stands for, how can i set the range, i have looked through the config file, there is no setting for range. Any bugs or script behind??

grotesk86, 2018-04-02:
what did you set for `position_endstop: ``position_max: `?

KevinOConnor, 2018-04-03:
For assistance with an issue, you must attach the klipper log file as described at:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Nume1977, 2018-04-04:
I was succefully printing calibration cubes trying to adjust the arm_length parameter until i get 20x20 size, but after a few runs i started to have this problem too!

Here is a copy of the log, when the config started to fail, and with a rollback. In both cases the X tower move order is higher than it's height! Resetting firmware and kippler does not solve this problem! I cannot print anymore xD

[klippy.log](https://github.com/KevinOConnor/klipper/files/1876997/klippy.log)

KevinOConnor, 2018-04-05:
@Nume1977 - that's by design.  When you home a delta the nozzle is going to be at the top of the printer.  You can't move the printer to Y50 at that Z height as attempting to make that move would cause carriages to crash into the top of the frame.  One must move the nozzle to a lower Z height before moving the nozzle horizontally.

KevinOConnor, 2018-04-18:
Closing this issue - post the log file if assistance is still needed.

closed 2018-04-18 by KevinOConnor

YaronSoffer, 2018-12-19:
My guess this is happing if you have done BED_MESH_CALIBRATION  ( my bed is not level at all).
somehow the calculation leads to a value of Z below zero. 
Here is an example of a failure :
Move out of range: 165.313 85.200 -0.531 
you can work around this by setting the following to negative value. I had set mine to almost minus one millimeter.
position_min: -0.99
my guess this can be done in the software ( taking the lowest value from the BED_MESH_CALIBRATION).