Klipper3d/klipper #287: Dual X-axis carriage configuration, Z axis not working

SrGeek, 2018-04-10:
I have my printer configured as a dual x axis carriage.  All axis home properly.  I can move axis X0,X1 and Y manually after homing.  After I home Z, I cannot move the axis manually.  Please see attached klippy.log file.
[klippy.log](https://github.com/KevinOConnor/klipper/files/1892741/klippy.log)

KevinOConnor, 2018-04-10:
The log is showing an "Endstop z still triggered after retract" error.  Double check that the endstops are working (as described in https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md ).  Also, see if the Z is retracting at all - if not - verify the Z direction pin.  If it's moving, but not moving enough, double check the Z step distance and Z homing_retract_dist parameters.

SrGeek, 2018-04-11:
Adjusted endstop and this solved the problem.  Thank you Kevin.

closed 2018-04-11 by SrGeek