Klipper3d/klipper #291: Z height after homing

Maximuscr31, 2018-04-11:
I have got all my motors going the correct way. I converted a Davinci 1.0 to Ramps 1.4.  My Z end stop is at the top of the screw axis. If I set it to pin ar 18 it homes the correct direction and end stops activate. Problem is this doesn't put my bed at the right height. It is above my nozzle. If I set the Z to 200 and 200 is the end stop position I need the bed to drop to 193.10 to be perfect. With that being said I have no idea how to make this work or if I am going about this the backwards way. Here is my setup file. I basically need to know how to go about getting it to home to the right height after hitting the end stop. 
Thanks
Brandon

# This file contains common pin mappings for RAMPS (v1.3 and later)
# boards. RAMPS boards typically use a firmware compiled for the AVR
# atmega2560 (though other AVR chips are also possible).

# See the example.cfg file for a description of available parameters.

[stepper_x]
step_pin: ar54
dir_pin: !ar55
enable_pin: !ar38
step_distance: .0125
#endstop_pin: ^!ar3
endstop_pin: ^!ar2
position_endstop: 237
position_max: 237
homing_speed: 5

[stepper_y]
step_pin: ar60
dir_pin: ar61
enable_pin: !ar56
step_distance: .0125
#endstop_pin: ^!ar14
endstop_pin: ^!ar15
position_endstop: 217
position_max: 217
homing_speed: 5

[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .000390625
endstop_pin: ^!ar18
#endstop_pin: ^!ar19
position_endstop: 193.10
position_max: 200

[extruder]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
step_distance: .002
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: NTC 100K beta 3950
sensor_pin: analog13
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 260

#[extruder1]
#step_pin: ar36
#dir_pin: ar34
#enable_pin: !ar30
#heater_pin: ar9
#sensor_pin: analog15
#...

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog14
control: watermark
min_temp: 0
max_temp: 130

[fan]
pin: ar9

[mcu]
serial: /dev/ttyACM0
pin_map: arduino

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

# "RepRapDiscount 2004 Smart Controller" type displays
#[display]
#lcd_type: hd44780
#rs_pin: ar16
#e_pin: ar17
#d4_pin: ar23
#d5_pin: ar25
#d6_pin: ar27
#d7_pin: ar29

# "RepRapDiscount 128x64 Full Graphic Smart Controller" type displays
#[display]
#lcd_type: st7920
#cs_pin: ar16
#sclk_pin: ar23
#sid_pin: ar17

akpch, 2018-04-11:
Tried "position_endstop: -193.10" yet?

KevinOConnor, 2018-04-11:
On Wed, Apr 11, 2018 at 03:25:17AM +0000, Maximuscr31 wrote:
> I have got all my motors going the correct way. I converted a Davinci 1.0 to Ramps 1.4.  My Z end stop is at the top of the screw axis. If I set it to pin ar 18 it homes the correct direction and [...]

The Z position parameters in the config are all relative to the
distance between the nozzle and bed.  So, if your endstop is near the
nozzle, then the position_endstop should be near zero.  If the endstop
triggers when the bed is above the nozzle, then you should set both
position_endstop and position_min to a negative number (eg, -6.9).

-Kevin

Maximuscr31, 2018-04-11:
Ok. Now that I do that it homes down instead of up towards the nozzle. I found the new master config. How do I use this command to make it chase 0 instead of 200?
homing_positive_dir

Maximuscr31, 2018-04-11:
Tried 
homing_positive_dir: False and it does not affect the direction
Tried 
homing_positive_dir: True and it won't Z home at all. I feel like the problem is glaring me in the face but I don't get it. Thanks for the help Kevin.

akpch, 2018-04-12:
I think what Kevin is saying you need to set both position_endstop and position_min to -193.1
It works.

Maximuscr31, 2018-04-12:
This was my last attempt. 
[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .000390625
endstop_pin: ^!ar18
#endstop_pin: ^!ar19
position_endstop: -6.9
position_max: 200
position_min: -6.9

 I took Kevin at that but will try -193.1 and see if it makes a difference.

KevinOConnor, 2018-04-12:
On Wed, Apr 11, 2018 at 07:14:28PM -0700, Maximuscr31 wrote:
> This was my last attempt. 
> [stepper_z]
> step_pin: ar46
> dir_pin: ar48
> enable_pin: !ar62
> step_distance: .000390625
> endstop_pin: ^!ar18
> #endstop_pin: ^!ar19
> position_endstop: -6.9
> position_max: 200
> position_min: -6.9

It sounds like you also need to invert the polarity of dir_pin, as
described in:
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md

-Kevin

Maximuscr31, 2018-04-12:
If I invert the polarity then my controls for the z axis in manual are backwards. I could change the wiring for my motor but that would still cause the same issue. It sounds like the software isn't capable of homing a z axis with the end stop at the top. I could move my endstop to the bottom but homing everytime I use it 200mm on a lead screw is going to be a nightmare.

KevinOConnor, 2018-04-12:
On Thu, Apr 12, 2018 at 02:53:27AM +0000, Maximuscr31 wrote:
> If I invert the polarity then my controls for the z axis in manual are backwards. I could change the wiring for my motor but that would still cause the same issue. It sounds like the software isn't [...]

I'm not sure what you mean by "my controls for the z axis in manual".
If you mean the OctoPrint control window, then go to OctoPrint
Settings -> Printer Profiles -> edit -> Axes -> select Invert control
for Z.

Homing the Z towards the nozzle is very common - likely the most
common setup.

All firmwares measure Z as distance from nozzle to bed, so you have to
configure the Z this way.

-Kevin

Maximuscr31, 2018-04-12:
Well that makes sense. I had no idea octoprint could invert axis controls. Thank you so much Kevin. I feel like an idiot but I hope this might help someone else. I will test it in the morning. Thanks again.

hg42, 2018-04-12:
For the software "down" means "lower values".
The inversion is caused by moving the bed instead of the nozzle.
If you imagine the nozzle would move instead of the bed, everything works like expected. It's like a camera mounted on the bed like it is in some printers.
If you invert octoprints axis controls, you might still be confused when directly using Z values (e.g. in slicer start code etc.) or may be some other software (e.g. Android app) that doesn't have this inversion.

closed 2018-04-18 by KevinOConnor