Klipper3d/klipper #292: Script to type in two number in config file

leungtech, 2018-04-11:
The bed tilt in klipper is ok , it can compensate skew bed, but it is very annoying to perform bed_tilt_calibrate and then type in xy number into the config file, then restart and print.I think it is possible to save xy number into the firmware and change config file automatically. I am not programmer but I think it is possible to do that great bed tilt function fully automatically after each bed_tilt_calibrate, we need script behind klipper. Can any programmer do that?Making kipper bed tilt just like marlin g29 auto bed leveling procedures. thx!

theopensourcerer, 2018-04-11:
You already asked this in #267, which was marked as a duplicate of #248.

Anyway - if I understand your question, this feature was already committed to master recently:

https://github.com/KevinOConnor/klipper/commit/5e060c3c65897c57705e90a8a4d66b9b6b5497ba

leungtech, 2018-04-11:
Good job, how can we use this script??do we need to update klipper？

KevinOConnor, 2018-04-11:
Duplicate of issue #248.

closed 2018-04-11 by KevinOConnor