Klipper3d/klipper #296: delta Z- offset dont work.

peed6687, 2018-04-17:
z_offset problems with delta

Hi, I built a Delta with Ramps 1.4 / TMC2100 @ Spredcycle 42.5v

A / B / C Motors and all sensors are runnin. For Z Min i Use a Piezo Sensor.
The Delta Calibration works fine, but I can not set an offset, positive and negative values do not get me anywhere, the distance nozzle to Heitzett is always at 0. Like the trigger point of the piezo. What am I doing wrong?

hg42, 2018-04-17:
you know you need to add a complete log file? it includes your settings and a lot more info

shadrincev, 2018-04-27:
Good afternoon. (I write using a translator, so may not be accurate) he Recently faced with this problem, its solution is simple: in [stepper_z] position_min: to specify a negative value, say -1. Then z_offset will be able to set the error to the nozzle in -0.3 or any other.

KevinOConnor, 2018-04-27:
Closing this issue pending a log file from the original submitter.

closed 2018-04-27 by KevinOConnor