Klipper3d/klipper #297: Z_offset and Z_position missing but its there in printer.cfg?

richardneo29, 2018-04-17:
[klippy.log](https://github.com/KevinOConnor/klipper/files/1920416/klippy.log)

Hi,

I have an anycubic kossel linear plus delta. When i start octoprint, it looks fine. based on logs attached, it says z_offset is missing. Therefore I rename probe_z_offset to z_offset. I did a RESTART and the next error turns up to be:

Error: Unknown option 'z_position' in section 'probe'
Unknown option 'z_position' in section 'probe'
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

Moreover to this, I am also getting an error I do not understand...

On boot up.. based on logs. it shows config error...

Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 235, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 211, in _read_config
    self.try_load_module(config, section)
  File "/home/pi/klipper/klippy/klippy.py", line 196, in try_load_module
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/probe.py", line 189, in load_config
    return PrinterProbe(config)
  File "/home/pi/klipper/klippy/extras/probe.py", line 19, in __init__
    self.z_offset = config.getfloat('z_offset')
  File "/home/pi/klipper/klippy/klippy.py", line 100, in getfloat
    minval, maxval, above, below)
  File "/home/pi/klipper/klippy/klippy.py", line 69, in _get_wrapper
    v = parser(self.section, option)
  File "/usr/lib/python2.7/ConfigParser.py", line 362, in getfloat
    return self._get(section, float, option)
  File "/usr/lib/python2.7/ConfigParser.py", line 356, in _get
    return conv(self.get(section, option))
  File "/usr/lib/python2.7/ConfigParser.py", line 340, in get
    raise NoOptionError(option, section)
NoOptionError: No option 'z_offset' in section: 'probe'
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.6.0-1-ga301713-dirty'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.9 (default, Sep 17 2016, 20:26:04) \n[GCC 4.9.2]'
Start printer at Tue Apr 17 13:00:35 2018 (1523970035.6 195.8)

richardneo29, 2018-04-17:
[klippy.log](https://github.com/KevinOConnor/klipper/files/1920425/klippy.log)

cmoeg, 2018-04-17:
please have a look in the extra-config file and you will find the missing information,...:
# Z height probe. One may define this section to enable Z height
# probing hardware. When this section is enabled, PROBE and
# QUERY_PROBE extended g-code commands become available. The probe
# section also creates a virtual probe:z_virtual_endstop pin. One may
# set the stepper_z endstop_pin to this virtual pin on cartesian style
# printers that use the probe in place of a z endstop.
#[probe]
#pin: ar15
#   Probe detection pin. This parameter must be provided.
#z_offset:
#   The distance (in mm) between the bed and the nozzle when the probe
#   triggers. This parameter must be provided.
#speed: 5.0
#   Speed (in mm/s) of the Z axis when probing. The default is 5mm/s.
#activate_gcode:
#   A list of G-Code commands (one per line) to execute prior to each
#   probe attempt. This may be useful if the probe needs to be
#   activated in some way. The default is to not run any special
#   G-Code commands on activation.
#deactivate_gcode:
#   A list of G-Code commands (one per line) to execute after each
#   probe attempt completes. The default is to not run any special
#   G-Code commands on deactivation.

KevinOConnor, 2018-04-18:
Your config has a "z_position" parameter in the probe section of the config file.  That's not a valid parameter - you should delete that line.

-Kevin

closed 2018-04-27 by KevinOConnor