Klipper3d/klipper #299: When re-plugging MCU into RPI after some time, /dev/ttyUSB0 can't be re-used

afoxinsocks, 2018-04-18:
[klippy.log](https://github.com/KevinOConnor/klipper/files/1925535/klippy.log)

At the tail end of this log file, which happened after many hours of not printing (I had turned off main power to the printer, and unplugged my MCU from USB):
1. I tried turning on 24v power to the MKS GEN 1.4
2. Waited several seconds, then plugged in USB to Raspberry Pi
3. Tried firmware_restart, to no avail (looking at the log file, it's because /dev/ttyUSB0 isn't there)
4. Looked at /dev, the MCU had enumerated as /dev/ttyUSB1
5. Replugged the board after an M112, and now the board enumerated as /dev/ttyUSB0 again
6. Tried firmware_restart again, it now successfully communicates.

Guessing file handle to the MCU isn't being closed when communications stop?

mmone, 2018-04-18:
Hi @afoxinsocks 

this might help: [FAQ / wheres my serial port](https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port)

afoxinsocks, 2018-04-18:
Wasn't using by-id. My bad.

closed 2018-04-18 by afoxinsocks