Klipper3d/klipper #300: Anet A8 don't start printing

Skudderon, 2018-04-20:
Hi@all,
i got Klipper running with my Anet A8 connected to a Raspberry Pi 3B with octoprint and Klipper-Addon.
I can connect to the printer, home all axes and move.
I can set temperature well.
I i start printing a code file the right temperature will set but after that nothing happens.
The A8 didn't start printing.
What can i looking for?

Best Regards
Henrik

[klippy.log](https://github.com/KevinOConnor/klipper/files/1932961/klippy.log)

Skudderon, 2018-04-21:
Dear Kevin,
Thank you for checking the logfile.
I was following this link and die the M112.
I will try again.

Henrik

Skudderon, 2018-04-21:
Thanks to you for your help. 
We found the problem in gcode generated by CuraEngine.
With S3D no problem.

closed 2018-04-21 by Skudderon

KevinOConnor, 2018-04-21:
On Sat, Apr 21, 2018 at 07:20:59AM +0000, Skudderon wrote:
> i was trying again.
> This time i also disabled the line wit M117 in code because i don't use a display.
> But there are no changes:

On a closer look, your Klipper version is several months old.  You
should upgrade to the latest version - see the FAQ for details:

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md

-Kevin