Klipper3d/klipper #301: Is the Anet A8 display functionally?

Skudderon, 2018-04-22:
Hi, i used the printer.cfg for Anet A8.
Works fine if display section is commented out.
If i activate this i get this message in terminal:

"Send: status
Recv: // Unknown command: hd44780_send_cmds
Recv: // This type of error is frequently caused by running an older
Recv: // version of the firmware on the micro-controller (fix by
Recv: // recompiling and flashing the firmware).
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Protocol error connecting to printer
Recv: !! Unknown command: hd44780_send_cmds
Recv: ok"

Klipper is up to date:
"Send: N1 M115*39
Recv: ok FIRMWARE_VERSION:v0.6.0-41-gbdd26e4 FIRMWARE_NAME:Klipper
What can i do?

Best Regards

dragonnn, 2018-04-22:
> Recv: // This type of error is frequently caused by running an older
Recv: // version of the firmware on the micro-controller (fix by
Recv: // recompiling and flashing the firmware).

Insert facepalm gif.

Skudderon, 2018-04-22:
So? Then tell me whats newer then 0.6.0-41???
*using same emoji*

dragonnn, 2018-04-22:
Klipper is up to date but the microcontroler part not. You need just to reflash you board like you did it first time installing klipper.

Skudderon, 2018-04-22:
I did it yesterday. The error was after update.

dragonnn, 2018-04-22:
Did you run make clean and make before flashing?

Skudderon, 2018-04-22:
No i didnt.
I used the way in:
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-upgrade-to-the-latest-software

dragonnn, 2018-04-22:
Then try this:
`make clean
make
make flash FLASH_DEVICE=/dev/ttyACM0`

Skudderon, 2018-04-22:
The make flash command does nit work with Anet board. So i used:
avrdude -p atmega1284p -c arduino -b 57600 -P /dev/ttyUSB0 -U out/klipper.elf.hex
like in the net A8 config file written.

dragonnn, 2018-04-22:
Doesn't matter, important is to use make and make clean to rebuild the firmware.

Skudderon, 2018-04-22:
"pi@octopi:~ $ make clean make make flash FLASH_DEVICE=/dev/ttyACM0
make: *** No rule to make target 'clean'.  Stop.!

dragonnn, 2018-04-22:
Not in one line. Every command is separate

Skudderon, 2018-04-22:
oh, ok thx

dragonnn, 2018-04-22:
And first you need to change folder with cd where you have klipper instaled

Skudderon, 2018-04-22:
THIS was the point ;-) thanks