Klipper3d/klipper #303: Progress and time doesn’t work on CR-10S Display

ChiliApple, 2018-04-23:
Hi,

I have activated the display functionality and the virtual SD Card with the paths in the .cfg file.
Display works fine, but the "%" and the time not.
the progress still stand @ "0" all the time, and the clock counter timed not right, sometimes he resets itself in print mode...

Thx, best Udo

ChiliApple, 2018-04-23:
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/1939690/klippy.log.zip)

KevinOConnor, 2018-04-23:
It doesn't look like the prints were using the virtual_sdcard mechanism.  Make sure OctoPrint is requesting an SD card print (in the "Files" tab on the OctoPrint web page, click the wrench icon, and select "Only show files stored on SD").  If you're not seeing any files in the OctoPrint list, try disconnecting and reconnecting in the "Connection" tab.

The reason the clock counter is resetting is due to "print stalls" - cases where OctoPrint is unable to send g-code commands as fast as the printer is processing them.  The fix for this is also to use virtual sdcard printing.

closed 2018-04-27 by KevinOConnor