Klipper3d/klipper #304: which config can be used with prusa mk3?

Crunch69, 2018-04-23:
which config can be used with prusa mk3?

KevinOConnor, 2018-04-27:
I don't know the answer to this question.  I suspect you'll need to figure out what electronics are in your printer to get further assistance.

Crunch69, 2018-04-29:
which config can be used with the Ensy Rambo that has the 2130 drivers

KevinOConnor, 2018-04-30:
Alas, the TMC2130 drivers are not currently supported.  Otherwise, the config/generic-mini-rambo.cfg example config is close to the Einsy Rambo.

-Kevin

RUWO, 2018-05-06:
when I asked ultralus if the 2130's were supported, he said they were in beta. I would just need to compile an alternate branch, and the branch was almost fully developed and functional according to him.

Elocai, 2018-05-10:
have the same board on a different  printer with same drivers of cource, would you pls ellaborate where the beta branch is that support Einy and 2130?

Elocai, 2018-05-10:
Shematics:
https://github.com/ultimachine/Einsy-Rambo/blob/0.5a/board/Project%20Outputs%20for%20Einsy_Rambo_0.5c/Schematic%20Prints.PDF

Electronics:
Drivers TMC2130 

@KevinOConnor tell me what information you need and I´ll provide

KevinOConnor, 2018-05-10:
If you wish to test, I've uploaded a sample config to the work-tmc2130-20180217 branch.  Pull the latest code on that branch and then use config/generic-einsy-rambo.cfg as a starting point for a config:

cd ~/klipper ; git fetch ; git checkout origin/work-tmc2130-20180217 ; sudo service klipper restart

The tmc2130 support on this branch is very basic (only spread cycle is supported, and not many parameters can be configured).  Limit testing to basic movement and functionality tests.

Elocai, 2018-05-11:
@KevinOConnor 

movement and endstop work, temperature readings ok

KevinOConnor, 2018-05-20:
FYI, basic tmc2130 support has been merged (commit 680c92de) as well as an example eisny config file (commit 2db5415d).

-Kevin

closed 2018-05-29 by KevinOConnor