Klipper3d/klipper #305: BED_TILT_CALIBRATE error

theopensourcerer, 2018-04-24:
Am tyring to understand and get bed tilt calibration and probing working. I have a home-built corexy (HEVO) with a Precision Piezo probe mounted to the tool-head. if I adjust the sensitivity of the probe too far so the probe is on and query_probe it reads TRIGGERED,  Otherwise it is open which is as it should be.

If I use the probe as an end stop it seem to work OK in that the bed stops moving up if I trigger the probe.

I was just trying to see how the bed tile calibration would work and got the following error in the Octoprint terminal and I attach a zip of the log file. (I think the error is around line 1099)

Thanks :-D

[image]

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/1943971/klippy.log.zip)

KevinOConnor, 2018-04-25:
This appears to be similar to issue #56.  (The log indicate the same, very puzzling, erratic ADC sampling.)  Alas, no solution was found for that issue.  Best guess was it was something odd going on with the atmega ADC - possibly due to high voltage.

Can you try again with the work-adc-debug-20180424 branch?  I'd like to see if that branch reports any shutdowns.

cd ~/klipper ; git fetch ; git checkout origin/work-adc-debug-20180424 ; sudo service klipper stop ; make flash FLASH_DEVICE=... ; sudo service klipper start

theopensourcerer, 2018-04-25:
@KevinOConnor Thanks for the follow up. 

FWIW I'm running the firmware on an el-cheapo Chinese MKS Gen v1.4 board (this one: https://www.aliexpress.com/item/BIQU-BIGTREETECH-GEN-V1-0-3D-Printer-controller-board-4-layers-PCB-MKS-Gen-V1-4/32792685922.html). I'm running it from a 24vDC supply which has been turned down to about 19v (to reduce current load to the heat-bed).

I also should note that the Precision Piezo Probe was extremely sensitive when I did this so the probe was triggering (bouncing) a lot, and very quickly, when the tool-head was moving. Maybe frequent changes on the Probe signal pin could be causing too many interrupts?

When I get home tonight I'll flash the branch above and run the BED_TILT_CALIBRATE command again and report back.

KevinOConnor, 2018-04-25:
On Tue, Apr 24, 2018 at 11:48:04PM -0700, Alan Lord wrote:
> Maybe frequent changes on the Probe signal pin could be causing too many interrupts?

FYI, Klipper doesn't use interrupts for endstops or probes (instead,
Klipper polls at regular intervals to detect a change).  So, too many
interrupts doesn't seem likely.

I guess it's possible that the frequent changes were causing noise
that somehow interferes with the ADC hardware, though.

-Kevin

theopensourcerer, 2018-04-25:
Same issue - happened very quickly this time. I'd adjusted the probe sensitivity so don't think it's the probe triggering frequency.

new log from the work-adc-debug branch:

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/1947881/klippy.log.zip)

KevinOConnor, 2018-04-25:
On Wed, Apr 25, 2018 at 09:17:43AM -0700, Alan Lord wrote:
> Same issue - happened very quickly this time. I'd adjusted the probe sensitivity so don't think it's the probe triggering frequency.
> 
> new log from the work-adc-debug branch:

I'm not sure what is going on.  As I suspected, the additional logging
indicates the ADC hardware got stuck for some reason.  Specifically,
the log indicates that the ADC is reporting busy (ADSC flag is set),
while no request has been made to it (last_analog_read = 0xff).

I can't think of anything that would cause this.  Even if the ADC
hardware were to spuriously take a sample, it should have finished in
~100us, but it was in this state for at least 25ms.

Did this work okay before using the probe?  If you simply disconnect
it, does the printer otherwise work?

-Kevin

theopensourcerer, 2018-04-28:
Just as a follow up - in case it helps - I "bit the bullet" this afternoon, worked out the configuration, and have flashed Marlin (1.1.8) onto my board. I just tried printing the same gcode file as above - it completed the print with no apparent errors.

I'm not sure what this really means, but I guess it validates the h/w and wiring of my printer to a certain degree at least.

KevinOConnor, 2018-04-28:
On Sat, Apr 28, 2018 at 03:18:15AM -0700, Alan Lord wrote:
> Hmm, don't know if this will help at all but my printer just stopped mid-print. See screenshot and attached klippy.log This was still running the debug firmware.

Okay, thanks.  Be aware that the debug firmware makes the issue more
likely to trigger - it's specifically looking for a delayed ADC
response.

That said, I never see even a delay of more than 2 ADC cycles - you're
getting a 200 cycle delay.  That's 25ms for what should take no more
than 100us.

On Sat, Apr 28, 2018 at 08:08:12AM -0700, Alan Lord wrote:
> Just as a follow up - in case it helps - I "bit the bullet" this afternoon, worked out the configuration, and have flashed Marlin (1.1.8) onto my board. I just tried printing the same gcode file as [...]

I'm not surprised that Marlin wouldn't report the issue.  Likely
Marlin would just wait a few hundred milliseconds when the event
occurs.  Unfortunately, that response leads to terrible print quality
with no hope of finding the root cause.

As near as I can tell, there is an hardware fault occurring in your
atmega2560 ADC (Analog to Digital converter).  I can't think of any
other reason for the chip to be reporting the ADC is busy for such a
huge amount of time.

Just to be sure, could you retry with the latest code from the
work-adc-debug-20180424 branch?

cd ~/klipper ; git fetch ; git checkout origin/work-adc-debug-20180424 ; sudo service klipper stop ; make flash FLASH_DEVICE=... ; sudo service klipper start

-Kevin

theopensourcerer, 2018-04-28:
lol - the part printed all the way to completion this time round. 

Attached is the klippy.log.

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/1958017/klippy.log.zip)

KevinOConnor, 2018-04-28:
Interesting.  All was not well on the chip, however.  See all the "ADC busy" messages in the log?  You shouldn't be seeing any of those messages.  (The latest debug code reports that if the ADC reports it is busy for more than 4ms.)

I'm wondering if there is a correlation between the heaters and the ADC (possibly due to an under-voltage situation).  Could you try setting heater_bed.max_power=0.4 and extruder.max_power=0.6 in the config?  Then run with the same debug code and see if you continue to see the "ADC busy" messages (and/or if they are less frequent).

-Kevin

KevinOConnor, 2018-04-28:
Also, if that doesn't help, and your heaters are okay with a 20ms switching time, you could try setting extruder.pwm_cycle_time and heater_bed.pwm_cycle_time to 0.020 in addition to the max_power settings.

theopensourcerer, 2018-04-28:
> I'm wondering if there is a correlation between the heaters and the ADC (possibly due to an under-voltage situation). Could you try setting heater_bed.max_power=0.4 and extruder.max_power=0.6 in the [...]

Not sure if that is the cause as the first time I ran into this issue I think the bed and hot-end were actually off. But sure - I'll do some more testing tomorrow. Thanks :-)

FWIW when I built this printer the combined current draw from _everything_ at 24v was going to be pushing the limit of my PSU (500W/20.8Amp at 24v) so I turned the voltage down on the PSU output to 19v - which in theory should reduce the current draw a bit too.

theopensourcerer, 2018-04-28:
> Also, if that doesn't help, and your heaters are okay with a 20ms switching time, you could try setting extruder.pwm_cycle_time and heater_bed.pwm_cycle_time to 0.020 in addition to the max_power settings.

What's the default? 0.010? Am not sure I understand the idea behind this suggestion? Care to explain when you have a mo - cheers.

KevinOConnor, 2018-04-28:
On Sat, Apr 28, 2018 at 01:10:51PM -0700, Alan Lord wrote:
> What's the default? 0.010? Am not sure I understand the idea behind this suggestion? Care to explain when you have a mo - cheers.

The default is 0.100 (100ms).  The idea is to reduce the amount of
consecutive time that the heaters are enabled.  (For example, at
max_power=.400 and pwm_cycle_time=0.020 the heater would be on for no
more than 8ms at a time.)

Additional switching could just as easily make things worse though (if
it has any impact at all).  So, I'd only test after seeing what the
impact of changing just max_power is.

-Kevin

KevinOConnor, 2018-04-29:
I just noticed the atmega2560 spec lists an unexpected interaction between the ADC and "Idle mode".  Can you grab the latest code (commit 61d070c6) and try with that (and with max_power / pwm_cycle_time unmodified)?

cd ~/klipper ; git fetch ; git checkout origin/work-adc-debug-20180424 ; sudo service klipper stop ; make flash FLASH_DEVICE=... ; sudo service klipper start

Thanks,
-Kevin

KevinOConnor, 2018-04-29:
FYI - I've now been able to reproduce this problem locally.  Further, the problem also goes away for me if I apply commit 61d070c.  So, it does appear to be the result of "Idle mode" (bizarrely, the atmega chips start an ADC conversion when you ask the chip to idle).

theopensourcerer, 2018-04-29:
Awesome. Thanks. I just printed the same item with the latest work-adc branch and I think there are zero ADC Busy messages :-) 

I see you pushed that commit to the master branch too so presumably I can switch back to that now then?
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/1958766/klippy.log.zip)

KevinOConnor, 2018-04-29:
Yes, the commit is now on the master branch (commit 4a8f9407), and yes, if you are no longer experiencing this issue then you should return to the master branch.

This was a nasty bug.  Thank you for reporting and testing.
-Kevin

theopensourcerer, 2018-04-29:
Nice one.

I'll close this one now then. Hopefully when I get a minute I'll get back to working out how to use BED_TILT_CALIBRATE with the Precision Piezo sensor.

closed 2018-04-29 by theopensourcerer