Klipper3d/klipper #306: Babystepping

ich777, 2018-04-26:
Hello,

is there any plan to add Babystepping (like in Marlin) for a live offset calibration if the Probe is not accurate enough in the future? I think it would be nice if you can do it on the Display or directly in Octoprint with an extra button or with a Plugin for Octoprint.

Regards

theopensourcerer, 2018-04-26:
I think what you are asking for is already done:

GCODE command SET_GCODE_OFFSET

https://github.com/KevinOConnor/klipper/issues/179

Functionality is described in the docs.

ich777, 2018-04-26:
Already seen that, but i mean a way to do it live on the display if you connected one or a way to add a button in Octoprint on the movement page for up and down with a specified value, or is there any Plugin for Octoprint where you could add buttons with a specific command?

ich777, 2018-04-26:
@theopensourcerer Tryed SET_GCODE_OFFSET, doesn't work

```
Send: SET_GCODE_OFFSET Z=-0.2
Recv: // Unknown command:"SET_GCODE_OFFSET"
```

theopensourcerer, 2018-04-26:
It was a very recent commit to the master branch. Have you updated your klipper code?

https://github.com/KevinOConnor/klipper/commit/6d03dee104531cc174b8ed8a0886768956bade31

ich777, 2018-04-26:
Installed it a few days ago, i will try to update and report back.
Thank you btw for the quick answers @theopensourcerer

mmone, 2018-04-26:
@ich777 In case you're interested I just added a dialog for the offset command to my [octoprint klipper plugin](https://github.com/mmone/OctoprintKlipperPlugin).

ich777, 2018-04-26:
@theopensourcerer Thank you, everything (now) works fine.

KevinOConnor, 2018-04-29:
We need the log to diagnose a report like this.  If a SET_GCODE_OFFSET command is not doing what you want, then issue an M112 immediately afterwards and attach the log file to this issue.  As described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

ich777, 2018-04-30:
@KevinOConnor sorry for the late response, tryed SET_GCODE_OFFSET Z_AUDJUST and SET_GCODE_OFFSET Z nothing worked for me or is something misconfigured?

[klippy.log](https://github.com/KevinOConnor/klipper/files/1961418/klippy.log)

mmone, 2018-04-30:
Just to make sure. You are aware that any offset is reset after a print is stopped or canceled?

ich777, 2018-04-30:
@mmone Yes, i've set this in the safe homing script that the offset is set to zero again.
My problem is that i can't audjust the offset at printing, in the documentation it says that you can use SET_GCODE_OFFSET Z at printing to set a fixed value or use SET_GCODE_ OFFSET Z_AUDJUST to increase or deacrease the offset.

KevinOConnor, 2018-04-30:
On Mon, Apr 30, 2018 at 11:14:06AM -0700, ich777 wrote:
> @mmone Yes, i've set this in the safe homing script that the offset is set to zero again.
> My problem is that i can't audjust the offset at printing, in the documentation it says that you can use [...]

Looking at this closer, it looks like the current Klipper
SET_GCODE_OFFSET command will not apply the change until the next move
that changes the given axis.  So, if you changed the Z, it wont take
effect until the next move with a "G1 Z..." parameter.

I'll have to think about how to fix that.

-Kevin

ich777, 2018-04-30:
First of all thank you for the quick reply and for the great software you put together.

Oh okay, i thought that it would take effect after the next movement gcode.
Okay, some other way planed to do a live offset change eventually without touching the gcode, like in marlin where you can live audjust the offset or after a movemnt gcode is finished?
Or is it possible to let's say you have a subcommand where you can change the Z offset without touching the gcodeoffset, i hope you understand what i mean, so that the printer or better the gcode dosen't recognice the offset is changed with something like a subcommand that overrides or comes bevor the gcode?

KevinOConnor, 2018-05-01:
Yes, I understand.  This looks like a bug in SET_GCODE_OFFSET.  I'll try to put together a fix for it.

ich777, 2018-05-02:
@KevinOConnor Thank you, my printer stands still until the fix is out, but that's not a big problem, need to make some cable managing and other stuff.

Again, thank you for this really cool, nice, usefull piece of software, like it very much and appreciate your work and the hours, days, weeks, month,... you spent on it to make it.

KevinOConnor, 2018-05-29:
I believe this should be fixed now (commit 1b1d2adb).

-Kevin

ich777, 2018-05-29:
Thank you, will test this tomorrow and report back.
Btw installed klipper on an Gigabyte Brix (Ubuntu Server 16.04.1 LTS, Intel i3, 6GB RAM, 120GB SSD) and works like charm, a little bit overpowered but the Brix is only sitting around in a corner.

ich777, 2018-05-31:
Issue can be closed, works now.

closed 2018-06-05 by KevinOConnor