Klipper3d/klipper #307: Extrusion speed

Skudderon, 2018-04-26:
Hi@all,
after your great support solving my first two problems klipper is running fine with my Anet A8.
Today i try to calibrate the printer and have 2 questions:
1. In Marlin i was setting steps; here is step_distance. How jan i calculate between?
2. For calibrating extrusion i try to extrude 10cm of Filament but if i do this in web interface i guess the extrusionspeed is too high. The extruder starts right but becomes more and more problems because of the counter pressure of the filament in the noozle. I changed from extruding 50mm twice to extruding 5mm 20 time waiting between some seconds but nothing changed. I extrude round about 4cm this way. 
How can i better calibrate extrusion?
Best Regards
Henrik

hg42, 2018-04-26:
step_distance is mm/step so
step_distance = 1 / steps_per_mm
or reversed
steps_per_mm = 1 / steps_distance

With a bowden you usually calibrate without the filament running through the hotend. 
If you have the hotend directly attached to the extruder, you probably need to use a slower extrusion speed if there is too much counter pressure. This should be adjustable in any host software. I currently use Printrun, so I don't know about Octoprint.

hg42, 2018-04-26:
for a first calibration you might also remove the nozzle and extrude without heating.
But I think Klipper will respond with an error, if you don't change some settings (not sure which exactly, min extrusion temperature etc.).

Skudderon, 2018-04-26:
Thanks, i will try this.

KevinOConnor, 2018-05-05:
I'm guessing you found your answer, so I'm going to close this issue.

closed 2018-05-05 by KevinOConnor

estebanavv, 2019-01-12:
Can someone elaborate how is the procedure to configure the e-steps (step_distance).
How is the 100mm converted to steps?