Klipper3d/klipper #309: All thermistor is "off" before homing

kevintong116, 2018-04-28:
I am using ramps fd v2.1 with due
After the machine is started, the temperature of all the thermistors become "off" before I manually home the printer.

If homing is not performed in a short period of time, I will get following error:

Recv: // Previous MCU 'mcu' shutdown: ADC out of range
Recv: // This generally occurs when a heater temperature exceeds
Recv: // its configured min_temp or max_temp.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! Previous MCU 'mcu' shutdown: ADC out of range

KevinOConnor, 2018-04-29:
We need the Klipper log to assist.  See: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

kevintong116, 2018-05-05:
[klippy.log](https://github.com/KevinOConnor/klipper/files/1976345/klippy.log)

kevintong116, 2018-05-05:
It's like if the motors are disabled, the thermistor will become "off"
[klippy.log](https://github.com/KevinOConnor/klipper/files/1976374/klippy.log)

KevinOConnor, 2018-05-05:
The log indicates that the temperature readings are very unstable.  There was also one spontaneous reboot of the Due as well.  As a guess, there might be a voltage problem with the Due and/or its sensors.

KevinOConnor, 2018-05-14:
I am going to close this issue pending further information.

closed 2018-05-14 by KevinOConnor

robustini, 2020-02-07:
@KevinOConnor same problem here, Ramps-FD V2, same ADC out of range.
The pin of both the thermistors are correct, verified with Marlin.
This board is very similar to RuRamps, which I am using without any problems with Klipper.
Tips?

[klippy.log](https://github.com/KevinOConnor/klipper/files/4171643/klippy.log)