Klipper3d/klipper #310: bed_tilt_calibrate crashing on bed with auto leveling sensor

andrebsb, 2018-04-29:
I'm having trouble using bed_tilt_calibrate.  The print head is crashing to the bed after I issue this command.
delta_calibrate is working fine and "query_probe" is responding as expected (triggered when I manually press it and open when not). 
But issuing a bed_tilt_calibration just crashes the head on the bed.  
i've reduced the points for the bed_tilt_calibration to only 3 as it would crash on the first one anyway. 
klippy log attached.
Thanks!

[klippy.log](https://github.com/KevinOConnor/klipper/files/1958463/klippy.log)

KevinOConnor, 2018-04-30:
I don't see any errors in the log.  One thing that looks unusual is your minimum_z_position = -50.85 setting.  This allows the nozzle to travel 50mm below the nominal location of the bed.  Typically, this value is not less than -2, but since your probe detection point is so far below the nozzle (15mm), I think this value should be zero on your printer.

As for the nozzle crashing into the bed during a bed_tilt_calibration operation.  Be aware that the initial move to the first probing point is a regular move.  So, the endstop positions need to be roughly correct in order to use bed_tilt_calibrate (and in order to use delta_calibrate).  To test this, one can run a regular G28 command followed by a "G1 Z10 F300" command - the nozzle should end roughly 10mm above the bed.  If this all looks okay, can you rerun the delta_calibrate command and forward the log file from that session?

-Kevin

KevinOConnor, 2018-04-30:
One other thing I just noticed - you have a horizontal_move_z=5 for bed_tilt_calibrate.  That value should be larger than the probe's z_offset (15mm).

-Kevin

closed 2018-05-14 by KevinOConnor