Klipper3d/klipper #313: thermal runaway protection, turn on all fans

cmock, 2018-05-01:
Hi,

the prusa i3 mk3, when a max_temp condition is encountered, turns on all fans at full blast.

The hotend cooling fan, if defined, should come on because the MCU goes into shutdown in case of overtemperature, I think.

Adding the part cooling fan to this may limit the temperature of the hotend in case the MOSFET fails, given the inability of some hotends to reach much above 200°C when it is on.

I think this would be a good safety feature to add to klipper.

(Credits go to Tom Sandladerer's recent video "Testing my printers for fire hazards")

grigorig, 2018-05-01:
The printer goes into shutdown mode when an issue with heating is detected. This should already switch on all fans.

grigorig, 2018-05-01:
Maybe I should clarify: Klipper has a "shutdown mode" which is entered whenever something out of the ordinary is detected. Timing errors, endstops not triggered when they should have been, temperature min/max error, or thermal control errors. Klipper configures shutdown mode to disable all heaters and to enable all fans by default.

KevinOConnor, 2018-05-01:
FYI, that is correct, but Klipper does not turn on the print cooling fan during a shutdown.  (It will turn on all the "heater_fan" fans during a shutdown.)  It's trivial to do that (the fan object even has a set_shutdown_speed() method), but I'm not sure there is much value in turning on the print cooling fan.

grigorig, 2018-05-01:
Well, the basic idea is that the layer fan always at least partially cools the hotend as well. So in the possible failure case that the hotend MOSFET fails and permanently switches on, keeping both fans active hopefully keeps the hotend from overheating.

Thanks for the hint, though. I was almost 100% sure that it was default for all fans to switch on in shutdown!

cmock, 2018-05-02:
I still think it's an added safety measure that has no drawbacks (that I can think of) for that "failed MOSFET" case. PLA has an auto-ignition temperature of 340°C, so everything one can do to keep the hotend as cool as possible would be good. 

Imagine a thermal runaway, where the part cooling duct made of PLA would then soften, stick to the heater block, and you have the conditions for a fire if those 340°C are reached. Also the nozzle, on shutdown, would probably be in contact with the print anyways. 

And given that E3D instructs to heat up to 280°C for tightening the nozzle, a standard 40W heater is apparently able to get dangerously near to that ignition temperature.

(I do have a heater block with stripped threads, so I could run an ignition test under controlled conditions next weekend if more data is needed)

KevinOConnor, 2018-05-02:
On Wed, May 02, 2018 at 01:53:46AM -0700, christian mock wrote:
> I still think it's an added safety measure that has no drawbacks (that I can think of) for that "failed MOSFET" case. PLA has an auto-ignition temperature of 340°C, so everything one can do to keep the hotend as cool as possible would be good. 

Well, in general, I prefer to turn things off during an error
condition then to turn them on.  (If nothing else, that's probably
safer in the event the wrong fan pins were defined in the config
file.)

That said, I've no objection to changing the code to enable the print
cooling fan during a shutdown.  Another option would be to keep the
current behavior but allow users to configure the shutdown speed (as
is done for output_pin).  Thoughts?

-Kevin

cmock, 2018-05-02:
Configuring the shutdown speed would make more sense in the grand scheme of things.

grigorig, 2018-05-03:
IMHO, if we can agree that this is a useful safety feature, then it should be enabled by default. Safety is not something you should have to "opt-in" to.

mmone, 2018-05-03:
In my opinion it makes no sense to have the fans turned on.
1. If a fire is developing the fans could provide more oxygen accelerating the combustion.
2. After a while the fans will melt down and most likely will create another short. With most fan outputs being able to deliver the same current then a heater output this might create a second ignition source.
If a runaway heater is cooled it might heat cycle at quite high temperatures instead of burning out in a relative short period of time thus increasing the chance that somethings starts to burn with a open flame.

Pure speculation on my side . Anyone feeling like sacrificing their printer in the name of science :-)

KevinOConnor, 2018-05-18:
I think in this situation the best thing to do is allow the user to specify the desired behavior (commit 2740838b).  The default continues to be to enable heater_fans and to disable print cooling fans on a shutdown.

closed 2018-05-25 by KevinOConnor