Klipper3d/klipper #314: Motorized autolevelling

pitucorto, 2018-05-01:
Hello,

Our 3Dprinter have four stepper motor for Z axis and the probe is the same extruder:

[image]

Here our config file [printer.txt](https://github.com/KevinOConnor/klipper/files/1963644/printer.txt)

We need to do some z height measurement and adjust each motor individually. Are there any way to do this without modify klipper code?

Congratulations for klipper, good work.

Regards.

KevinOConnor, 2018-05-01:
Several people have expressed interest in this recently.  This was raised previously in issue #283.  It does require some host code to be written.  My suggestion is to get everything up and running using the regular bed_tilt_calibrate command.  Once that is working well, if you're up for testing, then we can probably assist by providing some test code.

chainsol, 2018-05-01:
I will also be testing this - within the next week I'll either be writing the code myself or piggybacking off someone else's! I've got a Railcore that'll be done this weekend, with 3 Z steppers for leveling.

pitucorto, 2018-05-02:
We want do first test this weekend. We can test the code if you want.

Are any easy mode of move only one motor of the z axis? I have been reading the code of Klipper and I think for implement an autolevel the best is adding a module.