Klipper3d/klipper #317: Specify electronics/case cooling fan in config

w1ebr, 2018-05-02:
Hello Kevin,

I want to be able to specify a fan pwm output pin, a fan speed, a thermistor input pin and type and a threshold temperature (above which turns the fan on to the stated fan speed.) This would be used to cool my printer electronics.

Where do you recommend I start to code this in a way you will be willing to integrate into master?

Thanks!
Gene

theopensourcerer, 2018-05-02:
You _might_ be able to do this now... Take a look in the example_extras config file[1]. From line 194 onwards there are several examples of where you can define custom fans/heaters and various other digital pins. Maybe you could create a virtual custom "heater" that is always on (which would basically be the supply to your electronics) then you can create a thermistor and PWM pin definition for that?

Of course you'll need to have enough analogue input pins to support an extra thermistor, or perhaps you could use an external A2D, or something like a cheap I2C/SPI temp sensor device?

[1] https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L194

w1ebr, 2018-05-02:
Heaters are complex items that have ways to hold the proper temperature associated with them.

I am thinking that an extension of the [fan] definition (which is a cooling fan) could optionally  include a sensor and a threshold temperature. The issue with this approach is that Kevin's code assumes that sensors are associated with heating elements with closed loop feedback. 

Kevin, what do you think of an extension to the [fan] definition?

Gene

KevinOConnor, 2018-05-02:
On Tue, May 01, 2018 at 07:39:43PM -0700, w1ebr wrote:
> Hello Kevin,
>
> I want to be able to specify a fan pwm output pin, a fan speed, a thermistor input pin and type and a threshold temperature (above which turns the fan on to the stated fan speed.) This [...]

If you want to develop this, I'd say take klippy/extras/heater_fan.py
and copy it to something like klippy/extras/temperature_fan.py.  Then
modify the code so it instantiates a temperature sensor directly and
controls the fan as you need.

There's info on how to use the extras/ "module" system in the
Code_Overview.md document.

-Kevin

w1ebr, 2018-05-02:
Thanks, Kevin!

w1ebr, 2018-05-03:
I got so lost in trying to figure out what  functions are needed to define, set and get the temperature, as well as handle the thermistor look up, that I would now like to make this a feature request.

Thanks, Kevin!

w1ebr, 2018-05-04:
I am trying to understand the calling sequence for a case where a temp sensor is used

If I understand the heater class correctly, the INIT code is passed a sensor object but I can't find where the sensor class is defined.  

The heater class also contains the external functions needed to get the temperature and it uses a lock when it gets it. Do I need these functions defined in my module?

Thanks, Kevin.

Gene

KevinOConnor, 2018-05-04:
The sensor class is defined in klippy/extras/thermistor.py and klippy/extras/adc_temperature.py.  When they are run, they register themselves with the PrinterHeaters class (via the add_sensor() method).

If you're looking to create a new temperature sensor, I'd break out the sensor creation in PrinterHeaters.setup_heater() to its own method setup_sensor().  Then you can call `sensor = printer.lookup_object('heater').setup_sensor(config)` followed by `sensor.setup_callback(self.temperature_callback)` and define a temperature_callback method that copies the current temperature to a member variable.  The reactor callback (as seen in heater_fan.py) can then check that temperature to decide the fan speed.

The temperature_callback() method is invoked from a background thread, which is why the Heater class has locks.  However, for something this simple, it's possible to rely on python's basic thread safety mechanisms - copying the value to a local member (as described above) is atomic.

bmc0, 2018-05-05:
Here's a patch:  [add_temperature_fan.patch.txt](https://github.com/KevinOConnor/klipper/files/1976318/add_temperature_fan.patch.txt)

Needs documentation and more testing, but it seems to work.

w1ebr, 2018-05-07:
@KevinOConnor , I would like to push my modified version of heater and my new cooling_fan module to master. How do I arrange a code review with you?

Thanks!

Gene

KevinOConnor, 2018-05-08:
The M105 command is setup to report extruders and the bed - in general, trying to extend legacy g-codes is painful as g-code isn't flexible.

For contributing, see: https://github.com/KevinOConnor/klipper/blob/master/docs/CONTRIBUTING.md

-Kevin

KevinOConnor, 2018-05-10:
Several people have inquired on this support in the last few days.  @bmc0 - would you be willing to submit a pull request (or provide a signed-off-by line on your patch).  Or, similarly, @w1ebr if you have something ready?

Thanks,
-Kevin

w1ebr, 2018-05-11:
Hi @KevinOConnor ,

I have refrained from submitting it for a few reasons:

1) Not tested yet - I just got the hardware I need to test it with today. Tomorrow I should be able to test it.
2) I am not familiar with git yet, so creating a patch is new for me.
3) **The biggest reason is that this is a derivative case of a much more valuable capability:**
       IF input_pin is_in_some_desired_state THEN output_pin = some_desired_output

We already have [output_pin] in config and a command that can assign a value to it. (This is a more generalized version of the [fan] section, I think). All we need is an [input_pin] section that lets us define a pin and the associated data type, and something like an [action] section that refers to the input, the output and defines the condition.  Variations on this could be an [action] section that defines the input pin and type, the condition and references the output pin.

Thoughts?

Gene

hg42, 2018-05-11:
yes, this could also cover some other sensors, e.g. simple filament run out switch (action=pause) etc.

In general it's an event system.
Events could also be triggered by internal conditions etc.

w1ebr, 2018-05-12:
That is exactly my thought.  I don't know how Kevin feels about implementing almost the same feature multiple ways. I  am not a big fan of multiple methods to accomplish the same goals if I have to keep all of them up to date. That's why I am looking for Kevin's feedback. If he likes the idea, we can define the syntax and capabilities now, then implement it incrementally. 

Gene

KevinOConnor, 2018-05-14:
I'm not really sure what you're proposing, so it's hard for me to give an opinion on it.  If you implement something I'll take a look at it.

In general, I view Python as our "extensible, configurable, event driven system".  That is, I suspect the complexity of building, maintaining, documenting, and learning an extensible event configuration system is going to be greater than implementing a python module.  So, in general, I would target the config file for common setups, and target python modules for anything more advanced or unique.

w1ebr, 2018-05-15:
I agree that implementing and documenting a flexible approach using the config file would be more difficult than writing a specific python module, assuming someone is able to write a python module that works with the existing software base. I believe the majority of users will be people who don't feel up to the task. If I am wrong, I agree with you, Kevin. If I am right, I think implementing something using the config file would give Klipper a flexiible feature that no other printer firmware I know of currently supports.

Gene

w1ebr, 2018-05-15:
Here is a very rough example of the capabilities I think would be worth adding. With these there should be the possibility to support things we haven't even thought about yet.

[klipper_idea_1.txt](https://github.com/KevinOConnor/klipper/files/2006190/klipper_idea_1.txt)

wizhippo, 2018-05-15:
At first glance can't say I'm I fan of this.  Feels wrong to be scripting in the config file.
I do however like the idea of defining inputs and outputs and wiring them to the modules such as a `pid` instead of the defining the pin in `heater` itself.  I think instead you should have to define both the sensor and the pin/output, then pass them to a module such as `pid` or 'BangBang'.  You could then use the `pid` module to control heaters , fans etc.  We have part of this but I believe we could make it more reusable.  On the other hand that does add complexity and for most users defining a `heater` is more intuitive then defining a `pid`.

KevinOConnor, 2018-05-15:
On Tue, May 15, 2018 at 12:31:16PM -0700, Douglas Hammond wrote:
>On the other hand that does add complexity and for most users defining a `heater` is more intuitive then defining a `pid`. 

I'm getting consistent feedback that users are finding it difficult to
create the config file.  As such, I think it is worthwhile to have a
little additional complexity in the code so that config sections more
closely follow how users envision their printer components.  That's
why [extruder] simultaneously describes the stepper motor, the control
algorithm, and the temperature sensor.

I'm sure there are other ways to approach this, but I would not
recommend adding additional complexity to the common case to handle a
handful of uncommon scenarios.

Cheers,
-Kevin

KevinOConnor, 2018-05-18:
I'm going to close this issue, as the initial request is now in the repo.

closed 2018-05-18 by KevinOConnor