Klipper3d/klipper #318: Feature request: Interrupt-able M109 and M190

megablue, 2018-05-02:
Hi Kevin, I wish M109 and M190 is not fully blocking that I can just override the temperature or cancel heating altogether. say, I sliced a file to print with 220C, but I changed my mind, I wanted to print at 195C, I can just override the M109 while it is heating. I think repetier is able to do that, and a closed source chinese board (Lerdge) is able to do that as well. implementing it for M190 is a bigger deal though... since bed usually takes longer to heat up.  S0 to cancel heating and proceed to next commands. 

Thanks!

theopensourcerer, 2018-05-02:
I'm not sure I understand this. I'm using Klipper and if I want to change temperature of the bed or hot end on-the-fly I can. I just adjust the temperature controls in Octoprint - and it doesn't make any difference if I am waiting for the thing to heat up, or mid-print, the change takes effect pretty much immediately. Maybe I am I missing something...

megablue, 2018-05-02:
> What I do in this situation is issue an M112 command (that's a special
command that both OctoPrint and Klipper know how to handle
out-of-order), and then reconnect in OctoPrint (not sure why it
disconnects), and then issue a FIRMWARE_RESTART to clear the shutdown
state.

alright, thanks! I still hope this feature can be implemented though.

KevinOConnor, 2018-05-05:
Alas, I don't see any way to implement it in g-code as (aside from M112) commands aren't sent to Klipper until Klipper acks the previous command, and M109/M190 can't ack until heating is complete.

ich777, 2018-05-11:
@megablue If you want to change the temperature to another than you sliced it you can do it in octoprint itself with the temperatur offset.

closed 2018-05-14 by KevinOConnor

stemaDev, 2018-12-08:
Couldn't the host (klipper) handle the loop? Sending M104, M140 to the printer, checking back every 1 second and only continue parsing the gcode when it's up to the temp? That way the host could still handle commands (not exactly gcode). M112 doesn't always work, I can't always reconnect, and even if I do, I can't always restart the fw. I see why this wouldn't be possible with octoprint sending the gcodes, but what about printing from virtual sd card?

dlitz, 2021-07-27:
In fact, OctoPrint can do this already, if the firmware supports it:

[image]

Marlin supports this via the [M108 ("Break and Continue")](https://marlinfw.org/docs/gcode/M108.html) command to interrupt an in-progress [M109](https://marlinfw.org/docs/gcode/M109.html) or [M190](https://marlinfw.org/docs/gcode/M190.html) command.