Klipper3d/klipper #320: Multiple Instances of Octoprint, One Using Klipper, and Symlink Issues.

shoreliner, 2018-05-05:
Alright, log is attached that captures I think what happens when the problem is "active". Let me explain what I am doing here.  At this point I use Klipper only on my Anycubic Kossel Linear Plus, as I like it for delta kinematics in particular.  I have no issues at all when using it on an RPi by itself.  However, it seems the Monoprice Maker Select Plus(Wanhao Dup. i3+/Cocoon Create), throws my symlinks and the klipper service upside down.   I have Octoprint configured as such:

Instance

Octoprint          Monoprice Maker Select Plus.  SYMLINKED via rules.d to /dev/IIIP
Octoprint2        Anycubic Kossel Linear Plus -- This is my Klipper-driven machine.  sym'd to /dev/KOSSEL
Octoprint3        M3D Micro Original/iME Firmware/M33 Fio Octoprint Plugin. sym'd to /dev/MICRO
Octoprint4        FolgerTech Acrylic i3 revB(ramps14/Marlin118). sym'd to /dev/FOLGER

I am having constant issues with connections being overridden.  I recently started from scratch because I couldn't resolve the issue -- I have 20+ years of *nix experience.   When I did my make flash on the Kossel, I used the Symlink.  I'm using the default /tmp/printer serial connection.  However as you can see in the log, it seems as if it's trying to directly access /ttyusb0.  Just FYI on why I did the symlinks as such;  The reason I did this is because it seems like the monoprice printer likes to change from /ttyusb0 to /ttyusb1 whenever it likes.  I suspect it's a bootloader issue.  I have taken the Micro and FolgerTech printers out of the mix and still have the same problem with the kossel.  In fact, It seems like the only way to connect to the Monoprice MSP is to shut down the Klipper service.. like it is trying to rebind over and over.  
[klippy.log](https://github.com/KevinOConnor/klipper/files/1976263/klippy.log)

shoreliner, 2018-05-05:
I just noticed this as well..  in the log, it seems that my symlink was resolved to its actual:

[mcu]
serial = /dev/ttyUSB0
pin_map = arduino

Since I can't attach it, my printer.cfg is below while where you can see it has been defined using the symlink as follows:

[mcu]
serial: /dev/KOSSEL
pin_map: arduino

Anyways. I just wanted to get this submitted to solve some more base issues.. I am going to see if I can figure out why it's happening right meow.  Thanks all.

# This file serves as documentation for config parameters of delta
# style printers. One may copy and edit this file to configure a new
# delta printer. Only parameters unique to delta printers are
# described here - see the "example.cfg" file for description of
# common config parameters.

# DO NOT COPY THIS FILE WITHOUT CAREFULLY READING AND UPDATING IT
# FIRST. Incorrectly configured parameters may cause damage.

#  Z height probe. One may define this section to enable Z height
#  probing hardware. When this section is enabled, PROBE and
#  QUERY_PROBE extended g-code commands become available. The probe
#  section also creates a virtual probe:z_virtual_endstop pin. One may
#  set the stepper_z endstop_pin to this virtual pin on cartesian style
#  printers that use the probe in place of a z endstop.
[probe]
pin:  ^ar18
#   Probe detection pin. This parameter must be provided.
speed: 20.0
#   Speed (in mm/s) of the Z axis when probing. The default is 5mm/s.
#   The Z position to command the head to move to during a PROBE
#   command. The default is 0.
z_offset: 0

#	!!!		Stepper Configuration		!!!

#  !---------------STEPPER A-------------!

[stepper_a]
step_pin: ar54
dir_pin: !ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^ar2
homing_speed: 50
position_endstop: 292.313580
angle: 210.339923

#   Length (in mm) of the diagonal rod that connects this tower to the
#   print head. This parameter must be provided for stepper_a; for
#   stepper_b and stepper_c this parameter defaults to the value
#   specified for stepper_a.

#   Distance (in mm) between the nozzle and the bed when the nozzle is
#   in the center of the build area and the endstop triggers. This
#   parameter must be provided for stepper_a; for stepper_b and
#   stepper_c this parameter defaults to the value specified for
#   stepper_a.
arm_length: 268.0

#  !---------------STEPPER B-------------!

[stepper_b]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
[... 122 lines cut ...]
#   The speed (in mm/s) of non-probing moves during the
#   calibration. The default is 50.

horizontal_move_z: 5
#   The height (in mm) that the head should be commanded to move to
#   just prior to starting a probe operation. The default is 5.

manual_probe: true
#   If true, then DELTA_CALIBRATE will perform manual probing. If
#   false, then a PROBE command will be run at each probe
#   point. Manual probing is accomplished by manually jogging the Z
#   position of the print head at each probe point and then issuing a
#   NEXT extended g-code command to record the position at that
#   point. The default is false if a [probe] config section is present
#   and true otherwise.

# Reprap "2004" display
[display]
lcd_type: hd44780
rs_pin: ar16
e_pin: ar17
d4_pin: ar23
d5_pin: ar25
d6_pin: ar27
d7_pin: ar29

mmone, 2018-05-05:
This could possibly help: [FAQ.md/wheres-my-serial-port](https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port )

shoreliner, 2018-05-05:
Thanks man -- that may actually work, didn't really think of taking that path because typically symlinks are resolved differently.

KevinOConnor, 2018-05-05:
Using the /dev/by-id/ mechanism is probably the easiest way forward.

FYI, the config file in the log is the configuration that klipper is using - I'm not sure why there is a discrepancy, but klipper did not find a config file with /dev/KOSSEL (that and the bed pid settings are different).

shoreliner, 2018-05-07:
This is very weird... so, I've had pretty good luck with using /dev/by-id symlinks, but i've found that if I reboot the pi and all of my octoprint instances, including Klipper start, the first instance(With the Monoprice Printer Running marlin) cannot be reached.  Continually get a serial connection error...  the moment I stop the Klipper service, it is able to connect.   I am going to try and capture a better log of the issue... what I do now is only start with my first Octoprint service at startup, then once it is connected I start Klipper + octoprint2 instance, it connects, and so on.

shoreliner, 2018-05-07:
I have used the .../serial/by-id/ pointers as my 'Flash Device', I have also created symlinks... the behavior is the same.  As I thought I understood, Klipper should be attaching to /tmp/printer... that's what I am really confused about, especially when stopping the klipper service immediately alleviates the problem.

KevinOConnor, 2018-05-14:
Any further updates on this?  I don't really understand the problem you are seeing.

shoreliner, 2018-05-15:
Hey Kevin; sorry, things have been a bit busy with family stuff lately.  By
switching to using /dev/by-id, I haven't had any further issues.  The
problem only exists on devices that may have multiple usb-host controllers
and switch between, say, /dev/usbTTY0 and /dev/usbTTY1.  This is covered in
the FAQ and should be closed.  The only thing I can think that would
clarify things here would be to putting the /dev/by-id/ instructions in the
installation file.   Thanks!

Cheers,

David

closed 2018-05-18 by KevinOConnor

shoreliner, 2018-05-31:
Any of you guys on the list -- can you let me know if you get this?  I seem
to be having some issues.

Hey Kevin and All;

I am keeping revolve in the vapor category as well for now.    My concern
is that Elias seems to be always spread quite then(I know most of us are,
but well... i mean REALLY thin.)   The Magna Screen 2, another his
creations, was quite successful on his fundraising campaign but units that
were supposed to arrive back in August 2017 have now finally shipped... and
that's just for the first 500 backers.  This brings me to why I brought
this up.  I don't see how, already being behind on a product that was
announced 2 years ago, the board release that is supposedly in the next few
months, .  Anyways, I'm just rambling... I feel like if the community
focused on projects like Klipper, which don't require a specialized piece
of hardware, we'd be able to progress into some of the areas that need
attention.

Oh, and  as for the magnetic beds; it's just a really really simple
solution(From Prusa's mind, of course) that is used on the Legit Prusa I3
Mk3.  It just allows you to pull your build surface off, bend it a little
bit, and slap it right back onto the bed.   Anyways, just let me know if
you'd like one.. sending out a round of testers to people in the
community.  Also -- I'm basically finished re-establishing my lab, work
environment, and proper programming environments... Is there anything on
the TODO that you need in particular?

I have this tackled and will be submitting soon: "Add documentation
describing how to perform bed-leveling accurately in Klipper. Improve
description of stepper phase based bed leveling. "

I am thinking of tackling this next: " Complete the host based simulator.
It's possible to compile the micro-controller for a "host simulator", but
that simulator doesn't do anything currently. It would be useful to expand
the code to support more error checks, kinematic simulations, and improved
logging. "

Unless you have anything else in particular that would be helpful to
address first.  Oh, and the parser.. Are you familiar with configobj?  I've
used it before for oracle db migrations/batch ops for producing
configuration files.  https://pypi.org/project/configobj/  I'm thinking if
this was used for parsing printer.cfg, we could improve the automation of
actions like calibration routines(over multiple iterations), tuning
pressure advance, etc... and it preserves comments.  It would also give
users that don't feel comfortable in the config file a less daunting method
to calibrate.  It may even provide flexibility down the line that isn't
even being considered yet.  Now, time to put my printing hat on for about
an hour and jump into code.

I hope that you're well!

Thanks,

David