Klipper3d/klipper #321: klipper not installing properly on debian stretch

RUWO, 2018-05-05:
i was trying to get klipper to install onto debian stretch, but i got this error: 
Command "/home/pi/klippy-env/bin/python2 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-latyoq/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-L4uQLq/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/klippy-env/include/site/python2.7/cffi" failed with error code 1 in /tmp/pip-install-latyoq/cffi/
any idea what might be the cause of this? am i missing a dependency?
or some kind of version incompatibillit? i know the basics of python and bash, in case you want me to try something out.

RUWO, 2018-05-06:
I can't include the klippy.log, for what I hope are obvious reasons.

RUWO, 2018-05-07:
this is a regular RPi, with the newest version of raspbian(the distro that Klipper was built for as far as I know)
it is the UI-less version, but that should not be an issue. I know Klipper was built for Jessie and not stretch, it is quite hard to find a Jessie download. also, octopi will most likely upgrade in the next few months, so it would be nice to see that Klipper supports it. one thing i have thought of, is that this is the install-octopi.sh file, but i don't have octoprint installed(i would like to print from a virtual sd)

hg42, 2018-05-07:
do you specifically want Raspian? I think the usual way to install the pi would be to use octopi.

RUWO, 2018-05-07:
octopi is raspbian with octoprint pre-installed, and I don't want octoprint

hg42, 2018-05-08:
I'm sorry, my brain (or smartphone or app) mixed two different issues...not sure why...