Klipper3d/klipper #323: Print halted due to "MCU shutdown: Move queue empty"

ich777, 2018-05-05:
Hello,
my printer halted with the error: "MCU shutdown: Move queue empty" print speed was set to 50mm/s, this was the first time that this error occured, printing not from a virtual sd, printed from Octoprint directly.

Also found a issue #175 but he printed from the virtual sd card.

[klippy.log](https://github.com/KevinOConnor/klipper/files/1977069/klippy.log)

KevinOConnor, 2018-05-06:
Ouch, it seems you've hit a bad series of bugs.  :-(

The reason for this error is due to the Klipper host software being unable to get a good synchronization with the micro-controller clock.  Your log processed with `scripts/graphstats -f` shows: 

[image]

In particular, every 10 minutes or so, the micro-controller clock estimate becomes unstable.

I think this may be due to a subtle interaction between clock querying and lcd display updates.  I've updated a fix for this (commit 640ff1ad).  Also, be sure to recompile and flash the micro-controller code (to pickup another import fix - commit 4a8f9407).

-Kevin

ich777, 2018-05-06:
@KevinOConnor No problem, i like to test software in depth and i also like that you or somebody here replies really quick (not like in other projects where the answer takes 1 week or longer).
I also think that klipper will become a big thing in the 3d printing community or will be the next big thing. ;)

Okay, so i can perform an update if i understand that right and everything will be applied.

By the way i've got an idea for a filament sensor with a HAL sensor or with a opto endstop, sould i post this here as an issue (btw this has absolutely no priority, but i think this will be easily integratable and you also can detect clogs of the nozzle).
Will also report back on issue #319 when the update is done.

Sorry for my poor english by the way... :P

KevinOConnor, 2018-05-07:
Yes - update following the directions in the FAQ: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-upgrade-to-the-latest-software
Be sure to recompile and flash the micro-controller.

If you're referring to a filament run-out sensor, there is already an issue open (#249).  If it's something else, then I'm not sure how best to raise it.

-Kevin

ich777, 2018-05-07:
After updating the error has disappeard, should i upload a new klipper.log if everything is fine?

For the filament sensor i'm thinking to use an opto endstop with a small printable wheel that is connected to the filament and turns if the filament is extruded, mesures the open and close state of the sensor so that you can even detect cloggs of the nozzle.

Should be easier to implement i think, you could check if the extruder is moving and compare that to how fast the sensor opens and closes the switch or is this difficult to implement?
Eventually with a value that can be adjusted for clearance (don't find a better word in english) that it doesn't triggers the pause everytime when the extruder looses one step or slips one time.

This is only a idea and you solve two (or even three) problems with this solution (filament runout, clogged nozzle and eventually slipping extruder).

The best way to raise it would be with the M600 (Advanced Pause in Marlin) if you would like to trigger it manually, or what do you mean exactly?

KevinOConnor, 2018-05-08:
Sounds interesting.  If you are looking to implement this, see the docs/Code_Overview.md file for information on how to add new modules.

-Kevin

ich777, 2018-05-09:
@KevinOConnor I think this issue can be closed, never had that error again.
Sometimes when i turn the printer on i get the message "MCU 'mcu' shutdown: Timer too close" and it doesn't connect to it but can be solved if i issue the command 'FIRMWARE_RESTART', don't know whats causing this and it happens random.

My coding skills are not the best but i will see what i can do... :D