Klipper3d/klipper #325: lost communication with MCU 'mcu' - how to find out whats the problem?

StefanMeGit, 2018-05-06:
Hello! 

I got some shutdowns on some random parts... nothing special. I printed the part this morning with no issues.

i deleted a lot of lines in the middle to make the log smaller (sign with "...SWA")
[klippy.log](https://github.com/KevinOConnor/klipper/files/1977992/klippy.log)

What is the issue here? Can find any hint... 

the last lines are:

```
Read 552222.542175: 'SET_VELOCITY_LIMIT ACCEL=1500\n'
Read 552222.548801: 'SET_PRESSURE_ADVANCE ADVANCE=0.25 ADVANCE_LOOKAHEAD_TIME=0.005\n'
Read 552222.551444: 'N53611 G1 X110.600 Y74.450 F7200*126\n'
Read 552222.553596: 'N53612 G1 X110.875 Y74.700 F7200*119\n'
Read 552222.554812: 'N53613 G1 X111.150 Y75.023 F7200*126\n'
Read 552222.555899: 'N53614 G1 X111.381 Y75.378 F7200*122\n'
Read 552229.900750: 'N53615 G1 X111.565 Y75.759 F7200*112\nN53616 G1 X111.698 Y76.161 F7200*124\nN53617 G1 X111.779 Y76.577 F7200*112\nN53618 G1 X111.781 Y76.602 F7200*121\nN53619 G1 X122.684 Y96.471 F7200*116\nN53620 G1 X123.206 Y96.471 F7200*113\nN53621 G1 X123.206 Y97.423 F7200*118\nN53622 G1 X123.630 Y98.196 F7200*112\nN53623 G1 Z11.180 F1002*33\nN53624 G1 E0.0000 F3000*48\nN53625 G92 E0*64\nN53626 G1 X123.630 Y96.047 E0.0709 F4800*11\nN53627 G1 X65.630 Y96.047 E1.9855*93\nN53628 G1 X65.630 Y71.260 E2.8037*82\nN53629 G1 X65.647 Y71.032 E2.8113*81\nN53630 G1 X65.698 Y70.817 E2.8186*89\nN53631 G1 X65.783 Y70.610 E2.8259*91\nN53632 G1 X65.896 Y70.428 E2.8330*84\nN53633 G1 X66.042 Y70.259 E2.8404*87\nN53634 G1 X66.213 Y70.117 E2.8477*91\nN53635 G1 X66.408 Y70.000 E2.8552*87\nN53636 G1 X91.492 Y57.458 E3.7810*86\nN53637 G1 X123.630 Y57.453 E4.8419*99\nN53638 G1 X123.630 Y55.304 E4.9128*109\nN53639 G1 X134.370 Y55.304 E5.2673*104\nN53640 G1 X134.370 Y98.196 E6.6832*98\nN53641 G1 X123.630 Y98.196 E7.0377*105\nSET_VELOCITY_LIMIT ACCEL=1000\nSET_PRESSURE_ADVANCE ADVANCE=0.25 ADVANCE_LOOKAHEAD_TIME=0.005\nN53642 G1 X123.210 Y98.616 F7200*127\nN53643 G92 E0*64\nN53644 G1 X123.210 Y96.467 E0.0709 F4800*15\nN53645 G1 X65.210 Y96.467 E1.9855*89\nN53646 G1 X65.210 Y71.244 E2.8181*86\nN53647 G1 X65.231 Y70.968 E2.8273*94\nN53648 G1 X65.297 Y70.687 E2.8368*88\nN53649 G1 X65.408 Y70.418 E2.8464*86\nN53650 G1 X65.557 Y70.178 E2.8557*87\n'
Read 552229.911796: 'N53651 G1 X65.747 Y69.958 E2.8653*80\nN53652 G1 X65.969 Y69.774 E2.8748*90\n'
Read 552229.912264: 'N53653 G1 X66.207 Y69.631 E2.8840*92\n'
Read 552229.916089: 'N53654 G1 X91.393 Y57.038 E3.8135*87\n'
Read 552229.923937: 'N53655 G1 X123.210 Y57.033 E4.8638*98\n'
Read 552229.928078: 'N53656 G1 X123.210 Y54.884 E4.9347*106\n'
Read 552229.929875: 'N53657 G1 X134.790 Y54.884 E5.3170*109\n'
Read 552229.932437: 'N53658 G1 X134.790 Y98.616 E6.7605*101\n'
Read 552229.933386: 'N53659 G1 X123.210 Y98.616 E7.1428*101\n'
Read 552229.934317: 'N53660 G92 E0*65\n'
Read 552229.935264: 'N53661 G1 E-2.0000 F3000*30\n'
Read 552229.936141: 'SET_VELOCITY_LIMIT ACCEL=1500\n'
Read 552229.937833: 'SET_PRESSURE_ADVANCE ADVANCE=0.25 ADVANCE_LOOKAHEAD_TIME=0.005\n'
Read 552229.939511: 'N53662 G1 X121.620 Y80.729 F7200*124\n'
Read 552229.940846: 'N53663 G1 E0.0000 F3000*51\n'
Read 552230.511654: 'N53664 G92 E0*69\n'
Read 552230.515811: 'N53665 G1 X122.096 Y80.683 E0.0158 F4800*12\n'
Read 552230.518400: 'N53666 G1 X122.562 Y80.577 E0.0316*107\n'
Read 552230.625039: 'N53667 G1 X123.010 Y80.412 E0.0474*106\nN53668 G1 X123.434 Y80.191 E0.0631*106\n'
gcode state: absolutecoord=True absoluteextrude=True base_position=[0.0, 0.0, 0.0, 50750.45039999783] last_position=[134.79, 98.616, 11.18, 50757.21089999783] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0183333333333 extrude_factor=1.0 speed=88.0
Stats 552235.1: gcodein=29353252 mcu: mcu_awake=0.064 mcu_task_avg=0.000221 mcu_task_stddev=0.000249 bytes_write=51904314 bytes_read=9645631 bytes_retransmit=11112 bytes_invalid=6 send_seq=1118712 receive_seq=1118706 retransmit_seq=1118712 srtt=0.008 rttvar=0.004 rto=3.200 ready_bytes=206 stalled_bytes=3035 freq=15991639 heater_bed: target=100 temp=100.8 pwm=0.000 print_time=37511.841 buffer_time=0.004 print_stall=0 extruder: target=250 temp=249.1 pwm=0.604
Lost communication with MCU 'mcu'
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
```

khseal, 2018-05-08:
Check the Linux log. There may be interference with the USB if there is no ground. Connect the GND devices with a separate wire.

StefanMeGit, 2018-05-08:
Not saure what u mean with LINUX LOGs... u mean take a look into var/log?

And what do you mean with "Connect dthe GND devices with a separate wire? My pi is connected with an external power supply (MicroUSB Chager with 2A). Maybe disconnect the 5V part in the USB cable to the MKS Board?

KevinOConnor, 2018-05-08:
These "Lost communication with MCU" errors are typically due to some kind of system level or hardware event.  The host software stopped receiving any data from the micro-controller for an extended period of time (5+ seconds).  Things to look at would be the usb cables, the voltage levels (in particular, make sure the rpi power adapter is powerful enough), electronic noise (eg, check for proper grounding), etc.

Another thing to look at is the system logs in /var/log/  - there may have been a system event that occurred at that time.

-Kevin

nowerewolf, 2018-05-08:
I will also add, after having tons of issues with this... the raspi power has to be very very stable otherwise you'll get 'Lost communication'

closed 2018-05-18 by KevinOConnor