Klipper3d/klipper #326: Can't get fast retraction to work

TinHead, 2018-05-07:
Hi Kevin,

I'm pretty sure this is related to how the extruder speed is calculated and I did lots of printing without issue, but I think I've found a corner case:

I'd like to be able to set retraction speed, as it is currently Klippy retracts at it's own calculated speed as far as I can tell. 

This works great for most prints but for some like this one it failed for me as I can't tune the retraction speed (bowden setup):

https://www.thingiverse.com/thing:912478

I'm attaching the log for a test I did with the infaimous voronoy laser cat: 
[klipper.zip](https://github.com/KevinOConnor/klipper/files/1980958/klipper.zip)

And if I did not mixup files I think this was the gcode used:

[CFDMP_Laser_Cat_-_Voronoi_coarse.zip](https://github.com/KevinOConnor/klipper/files/1980967/CFDMP_Laser_Cat_-_Voronoi_coarse.zip)

Hope this makes sense.
Thank you,
Razvan

shoreliner, 2018-05-08:
> the only thing klipper does on its own is pressure advance if you activated it, one of its benefits is you need a lower rectration but its should not interferiere with your retraction settings

It would seem that, since klippy processes gcode before editing, removing things like jerky movements(like the zig-zag fill) and smoothing out changes in acceleration.... so, I'm a bit confused... klipper does much more than pressure advance.  Pressure advance is meant to assist at printing at very high speeds to reduce ooze and cornering issues.

TinHead, 2018-05-08:
Pressure advance is not used in this case.

KevinOConnor, 2018-05-08:
The klipper config specifies the maximum extrude only speed.  It's the slicer that determines the precise speed for each particular move.  So, you can increase the max speed in the config, and then set your desired speed in your slicer.

-Kevin

TinHead, 2018-05-08:
Exactly my problem, the slicer setting is ignored with both cura and slic3r ... ohhh you mean this: 
#max_extrude_only_velocity:

:man_facepalming: