Klipper3d/klipper #328: "Failed to home x: Timeout during endstop homing"

jtengland, 2018-05-09:
I had a print job fail on me about 82% into a 7 hour print. When I checked on it, the hotend had embedded itself into the top of the model as if it tried to home Z. An error was posted in OctoPrint that the reported error was "Failed to home x: Timeout during endstop homing." Sure enough, on line 23860 of the klippy.log, there was a random homing timeout, and the following lines show the temps dropping.

Thanks for any help!

[klippy.log](https://github.com/KevinOConnor/klipper/files/1986834/klippy.log)

jtengland, 2018-05-09:
Btw, this is occurred on a RasPi 3 and a CR-10 S5.

KevinOConnor, 2018-05-10:
Very strange.  There are no indications in the log of an internal Klipper error.  As near as I can tell, the request to home the X axis came from OctoPrint.  Oddly, though, before the X axis home timeout message, there is no g-code input from OctoPrint for ~45 seconds.  This is unusual, as even when not printing, OctoPrint normally sends temperature query commands every few seconds.  

Can you attach the g-code file that you were printing at the time?

jtengland, 2018-05-10:
Thanks for the response! 

Strange indeed. Sure, gcode attached.

[x9y19z18.txt](https://github.com/KevinOConnor/klipper/files/1993562/x9y19z18.txt)

mmone, 2018-05-11:
This could be related to the unfixed race condition in the current version of octoprint Gina Häußgen talks about it here: https://www.youtube.com/watch?v=Fq2Dl1Ge39c&t=213s

KevinOConnor, 2018-05-14:
Nothing in the gcode file looks odd either.  I've verified that Klipper can successfully parse the g-code in batch mode.  As near as I can tell, something upstream of Klipper did something wrong.

closed 2018-05-18 by KevinOConnor