Klipper3d/klipper #329: Small question

ramezquitao, 2018-05-09:
Is there a forum to ask questions related to klipper use and installation, that are not "issues"?.
I know this is not the best place to ask this, but I didn't found a better one.

Thanks

theopensourcerer, 2018-05-09:
There is a an IRC channel on freenode and a group on Facebook.

theopensourcerer, 2018-05-09:
There is also a mailing list:

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

ramezquitao, 2018-05-09:
Is the mailing list working? I haven't been able to use it.

Ricardo Amézquita Orozco
Doctor en Ciencias - Física
Director de Investigación y Desarrollo

CI Hologramas - Combustión Ingenieros S.A.S.
Cll 25B # 85B - 76, Bogotá / Colombia
Tel: 57-1-2635232
Fax: 57-1-4296516
http://www.cihologramas.com

ramezquitao, 2018-05-09:
Yes it is. 

Sorry

KevinOConnor, 2018-05-10:
The best way is via the mailing list, IRC, or a github issue.  See: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Cheers,
-Kevin

closed 2018-05-18 by KevinOConnor