Klipper3d/klipper #330: Problem when compiling firmware

wmlex, 2018-05-10:
Hello. I'm trying to compile the firmware for the microcontroller on the **HP t5325** (Marvell ARM 88F6281) under the control of **Debian 9**.
When executing the **make** command, I get the following error.

`:~/klipper$ make`
`  Build Kconfig config file`
`  Creating symbolic link out/board`
`  Compiling out/src/sched.o`
 ` Compiling out/src/command.o`
`  Compiling out/src/basecmd.o`
`  Compiling out/src/debugcmds.o`
`  Compiling out/src/gpiocmds.o`
`  Compiling out/src/stepper.o`
`  Compiling out/src/endstop.o`
`  Compiling out/src/adccmds.o`
`  Compiling out/src/spicmds.o`
`  Compiling out/src/pwmcmds.o`
`  Compiling out/src/lcd_st7920.o`
`  Compiling out/src/lcd_hd44780.o`
`  Compiling out/src/avr/main.o`
`  Compiling out/src/avr/timer.o`
`  Compiling out/src/avr/gpio.o`
`  Compiling out/src/avr/watchdog.o`
`  Compiling out/src/avr/serial.o`
`  Compiling out/src/generic/serial_irq.o`
`  Building out/compile_time_request.o`
`Version: v0.6.0-61-gf70fefa-20180510_073612-print`
`  Linking out/klipper.elf`
`lto1: internal compiler error: Segmentation fault`
`Please submit a full bug report,`
`with preprocessed source if appropriate.`
`See <http://gcc.gnu.org/bugs.html> for instructions.`
`lto-wrapper: avr-gcc returned 1 exit status`
`collect2: error: lto-wrapper returned 1 exit status`
`Makefile:87: recipe for target 'out/klipper.elf' failed`
`make: *** [out/klipper.elf] Error 1`
`make: *** Deleting file 'out/klipper.elf'`

`Linux print 4.9.0-3-marvell #1 Debian 4.9.30-2+deb9u5 (2017-09-19) armv5tel GNU/Linux`
`gcc-arm-none-eabi 15:5.4.1+svn241155-1 `
`gcc-avr 1:4.9.2+Atmel3.5.3-1`

KevinOConnor, 2018-05-11:
Alas, there are too many Linux distributions to test them all.  The build will work fine if you use the environment described in https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md .

Otherwise, the above indicates a bug in the Debian version of their arm compiler - I'd try upgrading to a new version and/or reporting it to Debian.

khseal, 2018-05-12:
As an alternative, you can compile the firmware on another device and flash the microcontroller.
I've been doing this for a long time, since on the old laptop there are no packages for compiling arduino duo.

closed 2018-05-18 by KevinOConnor