Klipper3d/klipper #332: disconnection from the MCU because of ADC

userosos, 2018-05-10:
Sometimes I have disconnection from the MCU because of the negative value on the terminal. In marlin it will be ok therefore, I think that this is not a hardware problem.
[klippy.log](https://github.com/KevinOConnor/klipper/files/1992602/klippy.log)

KevinOConnor, 2018-05-11:
If you are getting an MCU out of range error, then you can reduce the min_temp (and/or increase max_temp) in the config to avoid that.

However, I would not recommend doing that if the incorrect temperatures are due to external events as it weakens an important safety feature.  You may wish to consider gracefully shutting down the software before shutting down the hardware, or accepting the need to issue a FIRMWARE_RESTART at the start of your next session.

userosos, 2018-05-11:
@KevinOConnor 
It does not go beyond. In the configuration file, I set from 0 C to 280 C. Sometimes I see a reading of -70 C on the screen. If I set the minimum reading to -100 C, then the MCU needs a restart sometimes. This is not completely a problem in hardware, on MARLIN I have an oscillation of ~10 C (+- 5 C).

KevinOConnor, 2018-05-14:
If you set the min_temp to -100 and you need "a restart" then please post the klipper log file of the event as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

userosos, 2018-05-15:
@KevinOConnor i write-150 min_temp for extruder and hot_bed in configure file and it work! I has "Printer is ready" all times! Thanks for help!

closed 2018-05-18 by KevinOConnor