Klipper3d/klipper #339: Pause/Resume do not work, (i print from vSD)

ChiliApple, 2018-05-12:
Hy,

I can not pause / resume during a print (i print from vSD). Always get an error.
Use M24 / 25 M600 does not support, i want to change filament, a color change. 

thx, Best Udo

[klippy.log](https://github.com/KevinOConnor/klipper/files/1997633/klippy.log)

KevinOConnor, 2018-05-14:
I don't see anything in the log that indicates Klipper is doing something wrong.  The log indicates several pauses and resumes were implemented.  Some fail with a "Must home axis first" error.  Double check that you aren't turning off the stepper motors during the pause (and that there isn't a motor turn off (M84/M18) in the pause/resume scripts of OctoPrint - see the "GCODE Scripts" section of the OctoPrint Settings tab).

-Kevin

closed 2018-05-22 by KevinOConnor

ChiliApple, 2018-05-28:
Yes, it would be M18 codes inside, but it still does not work.
I also tried the position_recall.py script from wizhippo, but it still does not work.

With the M25 the printer pauses and goes to XY 0, after Resume the printer starts in the old position but with the Z position up (for one complete Layer, incl. extrusion) after that, he down Z to the right position.
But sometimes only Error in the Terminal "unknown GCode" SAVE_CURRENT_POSITION NAME
[klippy.log](https://github.com/KevinOConnor/klipper/files/2044731/klippy.log)

Am I the only one who uses a Pause / Resume function on a CR10? or maybe wants to change colors or just swap the filament?

Is there no way or solution for that?

THX best, Udo