Klipper3d/klipper #340: RaspberryPi is far ahead of print

StefanMeGit, 2018-05-13:
Is it normal, that the print in Octoprint is far far ahead? On an one hour print, Octoprint is 10 minutes earlier than the real print.

KevinOConnor, 2018-05-14:
It is not normal.  There's nothing further we can do without the log file - please attach it as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

closed 2018-05-20 by KevinOConnor

lucashpandolfo, 2018-06-12:
I have had this problem for some time, but only when engraving. Octoprint reports the work has finished way earlier than it should.
For the attached gcode file, Octoprint reports the job as finished with ```Print Time: 00:02:27```  but the real time is around 4 minutes or so.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2095350/klippy.log)
[example.gcode.txt](https://github.com/KevinOConnor/klipper/files/2095356/example.gcode.txt)

KevinOConnor, 2018-06-15:
So - it looks to me like OctoPrint doesn't like the massive amounts of SET_PIN commands.  I'd guess it doesn't expect an "ok" response from them, and thus doesn't do flow control properly for the rest of the print.  (That is, OctoPrint seems to be sending commands without pause to Klipper.)

For this particular use case, you could probably create some kind of G123 / G124 alias for your two SET_PIN commands so that they look like legacy g-code commands to OctoPrint.

-Kevin