Klipper3d/klipper #341: typo in stepper.py

ardvarticus, 2018-05-14:
I've noticed a typo in stepper.py causing Klipper to crash when the stepper phase end stop option is used, if the end stop accuracy specified is too low to make use of this feature.

137:            if self.homing_endstop_accuracy >= self.homing_stepper_phases // 2:
138:                logging.info("Endstop for %s is not accurate enough for stepper"
139:                             " phase adjustment", **name**)

The last operand on line 139 should be **self.name** rather than **name**. Correcting this typo prevents the crash and results instead in "Endstop for z is not accurate enough for stepper phase adjustment" being added to klippy.log

ardvarticus, 2018-05-14:
Log excerpt: 
=======================
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/scripts/install-octopi.sh/klippy/klippy.py", line 234, in _connect
    self._read_config()
  File "/home/pi/klipper/scripts/install-octopi.sh/klippy/klippy.py", line 212, in _read_config
    m.add_printer_objects(self, config)
  File "/home/pi/klipper/scripts/install-octopi.sh/klippy/toolhead.py", line 439, in add_printer_objects
    printer.add_object('toolhead', ToolHead(printer, config))
  File "/home/pi/klipper/scripts/install-octopi.sh/klippy/toolhead.py", line 236, in __init__
    self, printer, config)
  File "/home/pi/klipper/scripts/install-octopi.sh/klippy/cartesian.py", line 16, in __init__
    for n in ['x', 'y', 'z']]
  File "/home/pi/klipper/scripts/install-octopi.sh/klippy/stepper.py", line 206, in LookupMultiHomingStepper
    return PrinterHomingStepper(printer, config)
  File "/home/pi/klipper/scripts/install-octopi.sh/klippy/stepper.py", line 139, in __init__
    " phase adjustment", name)
NameError: global name 'name' is not defined
Internal error during connect.
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

KevinOConnor, 2018-05-14:
Thanks.  I committed a fix for this (commit 907cfb91).

-Kevin