Klipper3d/klipper #345: position_endstop setting position_min / position_max

HorseEars, 2018-05-15:
If an axis is setup with a `position_endstop` that is outside of the bounds of `position_min` or `position_max` the firmware will run and act normally, but the axis will be unable to home or function.
I think `position_min` and `position_max` should either be set to `position_endstop` if they aren't manually set on the axis, or prevent the firmware from running in this weird half functional state as it might not be very clear what is wrong. Especially as most of the board-specific example configs omit `position_min`

[klippy.log](https://github.com/KevinOConnor/klipper/files/2003268/klippy.log)
In this log, all axes are able to be homed except y.

KevinOConnor, 2018-05-18:
The goal of position_min and position_max is to specify the maximum range that the carriage should be able to travel.  So, I don't think it makes sense to specify a position_endstop outside of that range.  I improved the error reporting so this is more clear (commit 11816295).

Thanks,
-Kevin

hg42, 2018-05-19:
But homing isn't traveling, right?
When homing the position should generally be assumed unknown, because otherwise homing wouldn't make sense. No position => no check for min/max.

Taz8du29, 2018-05-20:
The problem arrived with manual probing (`BED_TILT_CALIBRATE`) for the `[bed_tilt]` section.
The values given by the firmware after calibration contained `position_endstop: -4.05.`,

I guess that an endstop out of range can happend when :
 - 3D printer has opto-endstops
 - Position 0,0 is acutally for the Z-probe, but a the same time, nozzle is out of the bed
 - user has a "parking zone" used to clean the nozzle easily

hg42, 2018-05-21:
@KevinOConnor so you see min/max as physical possible ranges.
My assumption was more like "while printing", because this is what I would want to have.
While printing, all coordinates should be kept inside this range, eventually clipping them at the limits.
Homing procedures should not be limited (because position should be invalidated).
And more advanced movements like parking a head (for changing it etc.) or wiping the hotend or pausing to change filament or engaging a z-probe by mechanical means etc. should be able to go outside the limits. This is because there can be disallowed zones for gemeral movements like printing, but they are reachable for special movements. Think of a delta, where you can easily reach outside the frame in certain directions.

E.g. for my corexy the hotend should always on the heatbed and berween the screws in the corners while printing. But homing goes outside the bed and the same for wiping the nozzle.

If these min/max values define the outmost limits, then I need additional limits for printing.

hg42, 2018-05-21:
also note, that the real limits cannot be described by a simple cube. There are usually several obstacles (e.g. the bed). But the printing area can be reduced to a cube (or may be a cylinder for a delta).
My hotend can easily be moved below the bed level when outside. Why should it be disallowed to move there e.g. to wipe the nozzle or similar.
Because of several obstacles, the limits cannot prevent crashing into them. So those limits are useless anyways.
But printing could be protected.

skuznetsov, 2018-05-22:
I found issue with probe is that it is moving too slow so it is not triggering endstop within given timeframe. When I am fiddling with position_min and setting that value to negative value it shows me incorrect Z value at bed level after probe triggered (I have inductive autolevel sensor ROKO SN04-N on  my Anet A8 clone) So at heatbed level it shows me Z -17 when I have my z position_min set to -30 and endstop_pos is set to 0.
PS: I found some other issue with probe but I will open another issue for that and some possible solution.

jakep82, 2018-05-29:
"E.g. for my corexy the hotend should always on the heatbed and berween the screws in the corners while printing. But homing goes outside the bed and the same for wiping the nozzle.

If these min/max values define the outmost limits, then I need additional limits for printing."

My corexy printer homes X at -3 so I have that defined as my X min and my endstop location.  My printer homes as expected with this setup, and my slicer keeps the printer within the limits of my bed by defining the print area as 0 to 200.  If I want to wipe my nozzle outside the bed area, I can use custom g-code to tell the printer to go to X -1 because that's defined as being within the physical limits of my printer.  It seems to me that everything you're describing is already possible.

theopensourcerer, 2018-05-29:
@jakep82 Agreed.

Mt CoreXY homes at X = -15 and Y = -45!

I've just added a wire brush to the side of the bed at X=240 to 280 and Y = -18 Y = -22 and written a small gcode_macro script to clean the nozzle on the brush:

> [gcode_macro clean_nozzle]
> gcode:  
>   G28  
>   G1 Z2  
>   G1 X280 Y-18 F3000  
>   G1 X240 Y-19 F4000
>   G1 X280 Y-20 F4000
>   G1 X240 Y-21 F4000
>   G1 X280 Y-22 F4000
>   G1 X240 Y-21 F4000
>   G1 X280 Y-20 F4000
>   G1 X240 Y-19 F4000
>   G1 X280 Y-18 F4000
>   G28

All works fine :-)

hg42, 2018-05-30:
@jakep82 @theopensourcerer your reasoning is missing the (=my) point.

"Everthing works" is valid even without min/max limits.

Why do we have limits at all? usually to prevent crashing into something, even if the slicer is configured wrong or you chose the wrong gcode file etc.

My printer can securely move within the corner screws of the bed.
Outside there are some things to obey.
End stops are outside, too.

Usually, if the firmware allows that, I use endstops for physical limits (kind of last resort, stopping the printer if any switch is triggered). This helps most of the time, when motors skip or with similar catastrophic failures (important for unattended prints).

Software limits are used for normal printing. This helps with most user errors.

For homing, software limits don't matter, because the position is unknown.

Wiping nozzles or similar actions after the print or may be changing hotends etc. while printing would need additional capabilities.

Several other firmwares can set limits at runtime, then there is no problem at all.
Allowing such settings to be changed at runtime would solve several of my remaining issues with Klipper.

Taz8du29, 2018-05-31:
> If you're getting the suggestion to use a negative position_endstop, it almost certainly means your probe's z_offset setting is incorrect.

I don't use any probe, I'm manually jogging the Z screw in order to calibrate (as described in [configs/example-extras.cfg][0]).

But as said above, my endstops are the last limit before damaging anything. That doesn't mean that I don't want to have my zero a bit above. Calibrating the offset from fimware is way easier than physically changing the endstop position each time (given that you may not even achieve a better accuracy than 0.3mm with this technique).

If you want another example: Let's say I change my nozzle, because the previous one is broken. but sadly, new one is longer. I don't want to change my endstop position, or even recalibrate my whole printer, in order to only have a Z offset of 3mm.

  [0]: https://github.com/KevinOConnor/klipper/blob/1b1d2adb31122648d8ba28fdbef9f4d2044cc5f5/config/example-extras.cfg#L58

TheAssassin, 2018-05-31:
Sounds like a really useful feature. Most other firmwares allow you to configure offsets for nearly _any_ detected value. Be it the origin, bed leveling stuff, etc.

I'd like to see some more flexibility here.

mfp20, 2020-03-14:
@hg42 I'm investigating the code for [another purpose](https://github.com/KevinOConnor/klipper/issues/2007) so I was trying to understand how the endstops are used by default in klipper, and [why there's the min only](https://github.com/KevinOConnor/klipper/issues/284), then [what is the lead developer idea about having min and max](https://github.com/KevinOConnor/klipper/issues/216#issuecomment-374016273), and finally I've stumbled on this "endstop klipper philosophy" thread.

As far as I can understand, klipper doesn't need the endstops to work, as everything is relative to software boundaries. It is kind-of-told in the docs. But as stated by Kevin in one of the issues linked above, we can implement the min/max endstops, as well as an extra module to add an extra check as you need; then ask for pull to the main repo.

More in general I share your concerns about safety, so I've installed redundant safety hardware. A big red "HALT" button included! And moved the safety logic outside the printer control boards, in order to skip any possible control software problem.
I've installed both min and max endstops on all axis (1 common x rail, and (z+y)*4 crane arms), as well as additional temperature/humidity/light sensors inside the printer box. All of those are (going to be) connected to a couple of cheap arduino clones, powered by independent dc-dc converters connected to the 2 independent PSUs (one for logic and steppers, 1 for hot stuff); then 2 heartbeats alert each other in case one of the two fail (ie: one of the 2 PSUs might have failed). One of those "monitor MCUs" is in charge of movement failures, the other one of heat control failures; both can pause, stop, turn-off one or both the PSUs (ie: turn off everything) using a couple of relays. This level of redundancy makes me feel comfortable with having half dozen 4-500°C heaters powered and moving on top of my wooden shelves (inside their fire-safe silicon/ceramic box). The software on the "monitor MCUs" is much simpler than the one running on the printer control MCUs; trivial looping on sensing and comparing the results to a static table; in theory the only unmanned failures can be a print fail (and filament/energy waste), or an AC loss; and those are not "catastrophic".
Currently I've the cabling in place, some parts mounted, and some parts still on my desk... so... can't say whether it is a good setup or not; but the whole cost for added safety is very low, and the system complexity not much higher.