Klipper3d/klipper #347: is ar1 a special pin?

w1ebr, 2018-05-15:
I can't get pin ar1 set to a value when I use either output_pin or temperature_fan in my config. Is there a way to see which pins are configured on the mcu?

This is a Rumba board and the pin is labeled PWM_1 and supposed to be available on the EXP3 header but the problem could be with the hardware.

Any suggestions on how to debug this?

Thanks!

Gene

cmock, 2018-05-16:
If you think it could be a hardware problem, write a short sketch in the arduino IDE and toggle that pin to see whether it does actually work?

w1ebr, 2018-05-16:
My problem was with pin numbering. I am now using chip pin names instead of Arduino pin names and all is good!

closed 2018-05-16 by w1ebr