Klipper3d/klipper #349: Request: Testing Klipper with Fabrikator Mini V2 / Malyan M100 (STM32F070 or maybe STM32F103)

3DSTB, 2018-05-17:
It would be very interesting to verify the possibility to use klipper with these machines. Unfortunitely i can not test it since i  don't have such printer. I'm not sure, if it is possible tp flash the firmware via USB or maybe over the SD Card (which is maybe not reversable?).
I already found some informations:
http://www.lemilica.com/reversing-mini-fabrikator-v2-malyan-m100-3d-printer/
http://wiki.malyansys.com/doku.php#firmware_for_m100
http://www.3d-druck-community.de/thread-17492-post-141215.html#pid141215
http://www.3d-druck-community.de/thread-17492-post-141323.html#pid141323
http://wiki.malyansys.com/doku.php#firmware_for_m100

It would be very nice to see a success.

KevinOConnor, 2018-05-22:
It should be possible to port Klipper to a wide variety of micro-controllers.  It's a bit of work to do each port though, so I suspect this wont happen in the short term.  If you're comfortable with embedded development, feel free to port the code and submit it.

Cheers,
-Kevin

3DSTB, 2018-05-24:
Hi Kevin,
thank you for your reply. There is another person working with the printer. I just asked him:
https://github.com/goran-mahovlic/stm32Board/issues/1

Regards
Steffen

KevinOConnor, 2018-06-01:
Interesting.  It will require someone with the hardware, development expertise, and desire to do the work necessary to bring it up though.

-Kevin

KevinOConnor, 2018-06-30:
It doesn't appear there is active development on this, so I'm going to close this for now.

closed 2018-06-30 by KevinOConnor

buzztiaan, 2018-07-03:
fyi, the 'monoprice mini select' (aka malyan m200) uses the same board, that malyan calls 'mm32'

there is a http://wiki.malyansys.com/doku.php?id=developer_guide_for_mm32_platform

tablatronix, 2019-08-30:
Did someone pick this up elsewhere?

KevinOConnor, 2019-08-30:
Klipper now has good support for the stm32f103.  I suspect the "Malyan M200" would not be difficult to get running with Klipper.  However, I don't know of anyone who has attempted it.

-Kevin

buzztiaan, 2021-10-16:
if need be , i have a spare motherboard of this type available for someone to try implementing this