Klipper3d/klipper #350: Request: add several heater_bed sections for multiple sections heatbeds

aegelsky, 2018-05-17:
Hi Kevin!

Would be super cool to add heater_bed1..N support like you did with multiple steppers.
With config of it's temp like in steppers endstops - use some parent heater_bed config (sensor type, pid, etc) or override with some specific passed.
Idea is to have multiple sections heatbed (I have 4 sections in total, printer is big),
so it would be possible to set each section ON/OFF, params and temperature (like test it with 80,90,100,110C to see adhesion differences).
This is the last thing that stops me from using your super cool Klipper and this feature is not implemented elsewhere afaik.

Best regards,
Alexander Egelsky

KevinOConnor, 2018-05-19:
On Thu, May 17, 2018 at 02:49:26AM -0700, aegelsky wrote:
> Would be super cool to add heater_bed1..N support like you did with multiple steppers.
> With config of it's temp like in steppers endstops - use some parent heater_bed config (sensor type, pid, etc) or override with some specific passed.

I'm not sure I understand - you have four heaters with one thermistor,
or four heaters with four thermistors?

For the former, you may be able to implement that with a [multi_pin]
configuration (see config/example-extras.cfg for the details).  The
latter would require some code changes - but, it's not something that
I'll have time to look at.

Cheers,
-Kevin

aegelsky, 2018-05-20:
I have four bed sections with four thermistors (per each), each bed section can be different power,
so the only option is to control every one. Also would be great to be able to set different temperature per section. Can't see the way where even 4 completely the same beds could be controlled by one thermistor - this looks leading to fire badly when one section can reach crazy temperature while other is still cold.

Idea of naming was like here (example-extras.cfg#L147):

> `# Multi-stepper axes. On a cartesian style printer, 
> One may define any number of sections with a numeric suffix
> `# starting at 1 (for example, "stepper_z1", "stepper_z2", etc.).
#[stepper_z1]

**hg42**
> [stepper x_stepper]
[heater_bed bed_heater]
[mcu ramps]
not sure what you proposed here, didn't met this in any config here (am I missing anything?).

Also it looks to me that I have two options:
1. use additional extruders that your firmware supports, but connect them to bed. Setting temp and configurating looks like pita. Also not sure which fake pins to feed in stepper config, because there will be no such.
2. use external hardware pid like rex c-100. Another pita and not really "professional way".
3. :) hire someone who can code python. Do you know any and how much would it cost?

Best regards and thanks for quick answer,
Alexander

hg42, 2018-05-21:
@aegelsky sorry for not being clear, my comment was about what I would want for klipper. It wasn't about a solution for your problem, but related to it. I don't think it would be difficult to have multiple bed sections, but the naming (and access to it) doesn't allow it.
Though is at least one thing to be solved, that is how temperature is reported to the host.
Most host software will only handle one bed, so either the hosts have to learn something or klipper has to fake a single bed. For the latter ir's difficult, because I can imagine different situations each demanding a different handling (min, max, average).

closed 2018-06-30 by KevinOConnor